smartdevicelink/sdl_core #966: [SDL4.0]SDL does not send OnSystemRequest to app on second device

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
always

**Preconditions:**
SDL4.0 feature is enabled in .ini file, SDL and HMI are started. App app_1 is registered via 4th on first device,has already received OnSystemRequest(Query_apps)and sent query json.

**Steps to reproduce:**
1. Register app via 4th protocol on second device.

**Actual result:**
After app is registered and sends to SDL OnHMIStatus(FULL) SDL sends OnSystermRequest(Query_apps) notification to mobile app on second device.

**Expected result:**
SDL does not send OnSystermRequest(Query_apps) notification to mobile app on second device.
**Note:** For reproducing defect using ATF extract files from files_for_script_execution_onSysterm_request_on_second_device.tar.gz to ATF_build and execute script Expecting_OnSystermrequest_query_apps_onSecond_device.lua.

**Environment:**
SDL:
sdl_repository = "https://github.com/smartdevicelink/sdl_core/"
sdl_branch = "release/4.2.0", commit fc832c7a7ae
ATF:
Repository: https://github.com/smartdevicelink/sdl_atf/
Commit: 962e1eff8664d97af5e59ec4e22f39b9ea76f5eb
Please see attached ATF_logs.zip

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority High.
Feature is unable for second device in standard use-case.

HSavynetska, 2019-11-25:
The issue is relevant on the current develop branch
Environment:
test script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2101
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

ShobhitAd, 2021-08-19:
Unable to reproduce the issue on the latest develop commit (https://github.com/smartdevicelink/sdl_core/commit/ca0cdaf10bae555a7bb5bdec0026adcc5f317eb6)

closed 2021-08-19 by ShobhitAd