smartdevicelink/sdl_core #967: SDL forwards OnButtonEvent notification of CUSTOM_BUTTON to BACKGROUND App.

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Core, HMI started.
2. App is registered on HM and has HMIlevel BACKGROUND

**Steps to reproduce:**
1. Send from HMI OnButtonEvent(CUSTOM_BUTTON, SHORT/LONG)

**Actual result:**
SDL resend it to BACKGROUND App.

**Expected result:**
SDL should resend OnButtonEvent notification to either FULL non-media App or FULL/LIMITED media App (see https://adc.luxoft.com/contour/perspective.req?projectId=65&docId=209491)

*Note:* 
to reproduce can be used lua script - attached.
for run:
* create folder user_modules in <ATF-2.2> folder.
* copy content of https://adc.luxoft.com/svn/APPLINK/doc/technical/testing/TestCases/lua/user_modules/ to user_modules
* run ./start.sh ATF_OnButtonEvent.lua

**Enviroment:**
commit: a4c2702
OS: Ubuntu 14.04 x64
ATF: v. 2.2
transport: TCP

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

robinmk, 2018-05-16:
We do not think the same requirement should apply to CUSTOM-BUTTON and hence think this bug is invalid.
Eg: Lets say we provide the ability for apps to populate softbuttons on the app list (like a short cut to the most widely used buttons). Then we would not want to limit the button presses being sent to the app in BACKGROUND.

GetmanetsIrina, 2018-05-23:
@robinmk , please confirm the Expected result:

**Expected result:**
In case  OnButtonEvent(CUSTOM_BUTTON, SHORT/LONG) is allowed by Policy SDL must resend it to BACKGROUND App.

Requirement 
>In case there is a media application in FULL HMILevel and several other media applications in BACKGROUND and-or NONE HMILevel connected to SDL and subscribed for Tune Up/Down, Seek Right/Left HW buttons, SDL must transfer the OnButtonEvent notification from HMI to the media application in FULL only.

will be still actual for Tune Up/Down, Seek Right/Left HW buttons.

robinmk, 2018-07-10:
@GetmanetsIrina @AStasiuk 

> Requirement
In case there is a media application in FULL HMILevel and several other media applications in BACKGROUND and-or NONE HMILevel connected to SDL and subscribed for Tune Up/Down, Seek Right/Left HW buttons, SDL must transfer the OnButtonEvent notification from HMI to the media application in FULL only.
will be still actual for Tune Up/Down, Seek Right/Left HW buttons

This still holds true for Tune Up/Down, Seek Left/Right buttons. The rule should be expanded to include Media applications that are in LIMITED as well. The behavior for LIMITED should be similar to applications in FULL.

Regarding CUSTOM_BUTTON, I would like to refer you to the description provided in HMI_API for the OnButtonEvent RPC. The description states rules that define the correct behavior.
[OnButtonEvent](https://github.com/smartdevicelink/sdl_core/blob/7f7fcbb998fb17f2954fd103349af67ea9b71a3f/src/components/interfaces/HMI_API.xml#L2704)

GetmanetsIrina, 2018-07-12:
Defect is reproduced.
SDL branch: develop(af97446)
ATF: develop
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1921

ValeriiMalkov, 2018-07-16:
The next link to PR which provided fix this issue:
[Pull request](https://github.com/smartdevicelink/sdl_core/pull/2365)

KrupenichOleg, 2018-11-12:
1) Description and summary updated according to comments written by @robinmk, this defect is valid now
2) Uploaded fresh script to reproducing this case (updated script from [PR#1921](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1921))

KrupenichOleg, 2018-11-12:
Defect is reproduced.
SDL branch: develop ([d363167](https://github.com/smartdevicelink/sdl_core/commit/d36316738785c96dab2ee892762ed08c059fffde))
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#1921

KrupenichOleg, 2018-11-13:
Defect is not reprodused 
SDL branch:  Bugfix `SDL forwards OnButtonEvent notification` ([3db5b52](https://github.com/smartdevicelink/sdl_core/pull/2365/commits/3db5b5237ee6e6f7b876182b3675213d43fe70fe))
ATF: develop
Script: [smartdevicelink/sdl_atf_test_scripts#1921](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1921)

LitvinenkoIra, 2019-12-10:
Defect is reproduced.
SDL branch: develop([e137556](https://github.com/smartdevicelink/sdl_core/commit/e137556512215ded94228e33d21ab4c3e1af3017))
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#1921

LitvinenkoIra, 2019-12-10:
https://github.com/smartdevicelink/sdl_core/pull/2365 - rebased to the latest develop 
Fix is valid

LitvinenkoIra, 2019-12-10:
Full regression successfully passed
SDL: fix/sdl_forwards_onbuttoneven c1b6be3a
Scripts: develop 6f43bfd
ATF: develop 37da7aba

LitvinenkoIra, 2019-12-10:
@theresalech Please triage this issue

theresalech, 2019-12-10:
@LitvinenkoIra thank you. We are currently working on finalizing the roadmap with the Steering Committee, so will keep this PR in consideration for upcoming Core releases.

iCollin, 2020-01-10:
Closed via #2365

PR #2739 (2018-11-06): Feature/button subscription response from hmi
Fixes #2508

This PR is **[ready]** for review.

### Risk
This PR makes **[major]** API changes.

### Testing Plan
ATF tests are [provided](https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2100)

### Summary
In this PR OnButtonSubscription notification is replaced with 
(Un)SubscribeButton request/response.

### Changelog
API was updated, as well as new request/response classes were introduces in SDL core.
Pending subscription resolution mechanism was introduced in case HMI responds to request after default timeout.
Moreover, ButtonNoti [... cut]
PR #2365 (2020-01-10): SDL forwards `OnButtonEvent` notification of `CUSTOM_BUTTON` to `BACKGROUND` App.
Fixes #967 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Summary
This PR provided bugfixing when SDL doesn't send notification `onButtonEvent` notification to mobile of `CUSTOM_BUTTON` in `BACKGROUND` app.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-10 by iCollin