smartdevicelink/sdl_core #972: HMI level resumption is not postponded at EmergencyEvent, isActive=true

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Core, HMI started.
2. Register media application.
3. Register non-media application.
4. Activate media application. HMI level of media becomes FULL. Add command to application.
5. Activate non-media application. HMI level media = LIMITED, HMI level non-media = FULL. Add command to application.
6. Perform suspend -> ignition off.
7. Perform ignition on. -> SDL and HMI are running.

**Step to Reproduce:**
1. Send OnEventChanged (EMERGENCY_EVENT, isActive=true)
2. Register media and non-media application.

**Expected result:**
SDL shall postpone HMI level resumption until receives OnEventChanged (EMERGENCY_EVENT, isActive=false)

**Actual result:**
SDL resumes HMI level of media and non-media application without receiving OnEventChanged (EMERGENCY_EVENT, isActive=false)

**Note:**
In SmartDeviceLinkCore.log OnEventChanged can not be found, so can be seen in ATF_Resumption.txt

**Enviroment:**
SDL branch: release/4.2.0
SDL commit: fa05b5f5b1af6fae7d684a5cc156ec2e68dbc197
HMI: HTML5
ATF version: commit 962e1e
OS: Ubuntu 14.04 LTS
App version: SPTv.2.1.73 Code 233
App type: Any

LitvinenkoIra, 2017-12-01:
Priority Medium.
Processing requests with wrong success value in case unavailable HMI component.

Ypostolov, 2018-06-05:
Not reproduced on "develop" branch.

Was fixed in 09b37dd3022f6e3bf12f323f14bd5c18f48caaf8.

AStasiuk, 2018-06-07:
Closed as not reproducible.

closed 2018-06-07 by AStasiuk

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/972_SDL_should_respond_IGNORED_with_correct_result_code_for_UnSubscribeVehicleData_in_case_vi_interface_is_not_available.lua