smartdevicelink/sdl_core #974: [API] SDL sends UpdateDeviceList with disconnected device in the deviceList

KhrystynaDubovyk, 2016-11-07:
**Occurrences:**
Always

**Precondition:**
SDL and HMI are started.

**Steps:**
1. Connect several devices.
2. Disconnect devices
3. Initiate sending UpdateDeviceList from SDL

**Expected result:** 
SDL sends UpdateDeviceList without disconnected devices.

**Actual result:** 
Disconnected devices are presented in UpdateDeviceList

**Note:**
To reproduce defect using ATF extract files for script execution in 'ATF_build' and execute script updateDeviceList_after_closingConnection.lua.

**Environment:**
SDL branch: release/4.2.0
SDL commit: fa05b5f5b1af6fae7d684a5cc156ec2e68dbc197
HMI: HTML5
ATF version: commit 962e1e
OS: Ubuntu 14.04 LTS
App version: -
App type: -

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

LitvinenkoIra, 2019-12-10:
Defect is reproduced.
SDL branch: develop(e137556)
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#1919

LitvinenkoIra, 2019-12-10:
https://github.com/smartdevicelink/sdl_core/pull/2249 - rebased to the latest develop
Fix is valid

LitvinenkoIra, 2019-12-13:
Need to rework the fix due to regression 

The following aft test scripts failed: 
./test_scripts/iAP2TransportSwitch/001_happy_path.lua 
./test_scripts/iAP2TransportSwitch/002_resume_failed_simple_case.lua
./test_scripts/iAP2TransportSwitch/003_resume_failed_remove_data_check.lua 
./test_scripts/iAP2TransportSwitch/004_commands_hold_while_switching.lua

SDL: https://github.com/Ypostolov/sdl_core/tree/fix/fix/update_device_list bba0d8d9
Scripts: develop 6f43bfd
ATF: develop 37da7aba

iCollin, 2020-01-22:
> Need to rework the fix due to regression
> 
> The following aft test scripts failed:
> ./test_scripts/iAP2TransportSwitch/001_happy_path.lua
> ./test_scripts/iAP2TransportSwitch/002_resume_failed_simple_case.lua
> ./test_scripts/iAP2TransportSwitch/003_resume_failed_remove_data_check.lua
> ./test_scripts/iAP2TransportSwitch/004_commands_hold_while_switching.lua
> 
> SDL: https://github.com/Ypostolov/sdl_core/tree/fix/fix/update_device_list [bba0d8d](https://github.com/smartdevicelink/sdl_core/commit/bba0d8d9c8be1e2f600a3f0b336a39a25faee3ed)
> Scripts: develop 6f43bfd
> ATF: develop 37da7aba

I see these passing, they require `-DBUILD_TESTS=ON`.

Closed via #2249

PR #2249 (2020-01-22): Fix UpdateDeviceList content
Fixes #974

This PR is ready for review.

### Risk
This PR makes no API changes.

### Testing Plan
unit tests
atf test

### Summary
fixed UpdateDeviceList contetnt  for case, when device connected
ower tcp protocol was unexpectedly disconnected and
device has only one connection.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-22 by iCollin