smartdevicelink/sdl_core #977: [SDL4.0]SDL sends appName in vrSynonyms and ttsName in case of lower and upper bound values of params in json file

KhrystynaDubovyk, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
SDL4.0 feature is enabled in .ini file, SDL and HMI are started.

**Steps:**
1. Register app via 4th protocol.
2. App sends via SystemRequest json file JSONWithLanguageDefaultVrTtsLowerBound.json
JSONWithLanguageDefaultVrTtsUpperBound.json

**Expected result:** 
SDL sends in BC.UpdateAppList values of vrSynonyms and ttsName from json file

**Actual result:**
SDL sends in BC.UpdateAppList in vrSynonyms and ttsName value of appName
Notes: To reproduce defect using ATF extract attached files in 'ATF_build/' folder and execute TtsNameVrSyn_lower_bound_language.lua.

**Environment:**
Tag: 4.2.0
commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF version: 2.2 / 962e1eff8664d97a 

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Low.
Defect is related to processing values of lower and upper bound.

ShobhitAd, 2021-05-12:
Closed via https://github.com/smartdevicelink/sdl_core/pull/2705

closed 2021-05-12 by ShobhitAd