smartdevicelink/sdl_core #982: SDL responds "resultCode: SUCCESS" while dataType:VEHICLEDATA_EXTERNTEMP is VEHICLE_DATA_NOT_AVAILABLE and not in subscribed list store

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Core and HMI are started.
2. SubscribeVehicleData (speed, externalTemperature), OnVehicleData is allowed by policy for app_1 and app_2
3. Navi App1 and navi app2 are registered and activated.

**Steps to reproduce:**
1. Send SubscribeVehicleData (speed, externalTemperature) from app_1
=> SDL responds SubscribeVehicleData ((speed:success),(externalTemperature:VEHICLE_DATA_NOT_AVAILABLE), SUCCESS)
2. Send SubscribeVehicleData (speed, externalTemperature) from app_2
3. Observe the result

**Actual result:**
SDL sends "resultCode: SUCCESS, success:true" to mobile app2.

**Expected result:**
SDL should send "SUCCESS, (speed: SUCCESS),
(externalTemperature: VEHICLE_DATA_NOT_AVAILABLE)" to mobile app2.

**Environment:**
SDL repo: https://github.com/smartdevicelink/sdl_core.git
SDL branch: master
SDL commit: a6c32b4
HMI: HTML5
OS: Ubuntu 14.04 LTS
Transport: WiFi
Mobile device: HTC One
Mobile OS: Android 4.3
Mobile App version: SPT v.2.1.81 Code: 239
Mobile App type (media/non-media/navi/any): any

**Attachments:**

JackLivio, 2017-11-07:
Issue is reproducible when App2 sends SubscribeVehicle data Speed:True and externTemp:True in the same RPC request.

jacobkeeler, 2017-11-30:
Closing with the merge of #1929

PR #1929 (2017-11-30): Fix/subscribe vehicle data cashing results
Fixes #982 

### Risk
This PR makes no API changes.

### Testing Plan
Verified with the attached script :  [982_atf_check.lua.zip](https://github.com/smartdevicelink/sdl_core/files/1507059/982_atf_check.lua.zip)
Verified on ATF version : a6a2a9b
Used atf scripts version : 38abfa9

### Summary
The root cause of the issue was an invalid type of keys for SubscribeVehicleData response.
Created local converter from VehicleDataType enum to string for converting already subscribed items. 
 
Also, this pull request contains removing enum VehicleDataType from  [... cut]
closed 2017-11-30 by jacobkeeler