smartdevicelink/sdl_core #985: [Policy]: SDL does not apply url from PT for specified appID for OnSystemRequest

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. SDL built with DEXTENDED_POLICY: OFF
2. Core, HMI started using attached sdl_preloaded_pt.json.
3. SPT is configured to send attached PTU - PTFromCloud_smoke%20test__2.json.

**Steps to reproduce:**
1. Register App with appID "584421907" - PTU trigered successfully.
2. Register app "Base", appID = 12345.

**Actual result:**
After Registration second app SDL triger PTU. SDL sends OnSystemRequest with url: http://policies.telematics.ford.com/api/policies

**Expected result:**
SDL send http://policifffffffffffff.com/api/policies url in OnSystemRequest notification to mobile.

**Environment:**
Release: 4.2.0 (master marge)
Repository: https://github.com/smartdevicelink/sdl_core
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
OS: Ubuntu 14.04 
Transport: WiFi 
App type: navi
App: Luxoft SPT
Device: HTC One 

**Attachments:**

JackLivio, 2017-04-18:
@DrachenkoAnastasiia Was this issue addressed in the 4.3 code at all?

JackLivio, 2017-08-01:
fix is merged

closed 2017-08-01 by JackLivio