smartdevicelink/sdl_core #986: SDL creates redundant device_consent table in Policy DB

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Core, HMI started 1st time.

**Steps to reproduce:**
1. Register App

**Actual result:**
SDL should add all device related info to Policy DB (according to dictionary identifier should be present)

**Expected result:**
SDL do not add all device related info to Policy DB

**Environment:**
SDL branch: master
SDL commit: a6c32b4
HMI: HTML5
OS: Ubuntu 14.04 LTS
Transport: WiFi
Mobile device: HTC One
Mobile OS: Android 4.3
Mobile App version: SPT v.2.1.81 Code: 239
Mobile App type (media/non-media/navi/any): any

JackLivio, 2017-04-18:
@KhrystynaDubovyk Do you know if this bug still exists or was fixed in the recent policies release?

KhrystynaDubovyk, 2017-04-19:
@Itileda 
Igor is this bug fixed?

dcherniev, 2017-05-16:
@JackLivio Checked this issue on the head of the develop branch, the device identifier is present in Policy DB, so this issue no longer occurs.
Checked on SDL commit: 458fa67468b2c10a7f8eb18277a83eb90c2f8004

JackLivio, 2017-11-07:
@KhrystynaDubovyk  @dcherniev When using an Sqlite Browser to examine the policy db after connecting an app and then  performing an ignition cycle, I see the the device identifier is present but all of the device fields are empty.

JackLivio, 2017-11-07:
Also is this issue for saving Device info or Device Consent groups? The title does not match the issue description. The title says there is redundant info saved, but the description says that no info is saved.

dboltovskyi, 2017-11-20:
@JackLivio 
Since the issue is quite old names of the tables has changed. Actual ones are "device" and "device_consent_group".
Both are related to device consent functionality and External Proprietary flow.

If SDL is built in default (PROPRIETARY) mode these ones are almost empty after app registration.
But if SDL is built in EXTERNAL_PROPRIETARY mode, and new app registered, tables became populated with appropriate data.

So for now this issue is not actual.

LuxoftAKutsan, 2017-11-20:
@JackLivio root cause of this defect is the same as in #1007. So it is not actual any more

closed 2017-11-20 by JackLivio