smartdevicelink/sdl_core #987: INVALID_DATA received in case wayPointType with correct parameter was sent

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**

**Preconditions:**

**Steps to reproduce:**

**Actual result:**
GetWayPoints_wayPointType_IsExistentValues_ALL_SUCCESS and
GetWayPoints_wayPointType_IsExistentValues_DESTINATION_SUCCESS
are failed
INVALID_DATA is received

**Expected result:**
SUCCESS should be received

**Environment:**
Release: 4.2.0 (master marge)
Repository: https://github.com/smartdevicelink/sdl_core
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
OS: Ubuntu 14.04 
Transport: WiFi 
App type: navi
App: Luxoft SPT
Device: HTC One 

**Attachments:**

jacobkeeler, 2017-03-28:
@DrachenkoAnastasiia The same occurs with `SubscribeWayPoints` and `UnsubscribeWayPoints`. I've found that part of the reason for this is that [CSmartFactory.h](https://github.com/smartdevicelink/sdl_core/blob/aff7f09a4ec07becfb64dc78992580061d3342ff/src/components/formatters/include/formatters/CSmartFactory.h#L289) is unable to find the Schema for these three RPCs, but I do not know why they are missing from the list.

jacobkeeler, 2017-11-07:
Despite there being several issues with the SDL HMI regarding this RPC (missing `code` and `appID` parameters in the `Navigation.GetWayPoints` response), this issue does not appear to be caused by anything in Core.

closed 2017-11-07 by jacobkeeler