smartdevicelink/sdl_core #994: [SDL4.0] Json validation is failed in case language parameter does not contain vrSynonyms or ttsName

KhrystynaDubovyk, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:**
 SDL4.0 feature is enabled in .ini file, SDL and HMI are started.

**Steps:**
1. Register app via 4th protocol.
2. App sends via SystemRequest json file JSONWithTtsNameVrSynonymsInSeparatedStructs.json

**Expected result:**
 SDL considers json file as valid and write a value of appName to the omitted param only.

**Actual result:** 
SDL fails json validation with messages:

> :ValidateLanguages: :QUERY_APPS_VALIDATION_FAILED: 'languages.ttsName' doesn't exist

**Notes:** 
To reproduce defect using ATF extract attached files in 'ATF_build/' folder and execute json_with_omited_ttsName_vrSynonyms.lua.

**Environment:**
Release: 4.2.0 (master marge)
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF: 962e1eff8664d97af5e59ec4e22f39b9ea76f5eb

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority High.
Not success request processing and json validation in standard use-case with valid all values and params.

jacobkeeler, 2018-08-29:
The provided json file is not valid given the validation criteria used by Core, it uses an array of objects with a single key each for language:
```
            "languages": [
                    {
                        "default": {
                            "ttsName": "Rock music App tts name default",
                            "vrSynonyms": [
                                "Rock music App 1 default",
                                "Rock music App 2 default"
                            ]
                        }
                    },
                    {
                        "EN-US": {
                            "ttsName": "Rock music App tts name EN"
                        }
                    }
                ]
```

It should just be an object with these keys:
```
                "languages": {
                     "default": {
                        "ttsName": "Rock music App tts name default",
                        "vrSynonyms": [
                            "Rock music App 1 default",
                            "Rock music App 2 default"
                        ]
                    },
                    "EN-US": {
                        "ttsName": "Rock music App tts name EN"
                    }
                ]
```

GetmanetsIrina, 2018-08-30:
@jacobkeeler , according to comment above regarding `language` structure was created the issue for scripts updating with structure
```
                "languages": {
                    "default": {
                        "ttsName": "Rock music App tts name default",
                        "vrSynonyms": [
                            "Rock music App 1 default",
                            "Rock music App 2 default"
                        ]
                    },
                    "EN-US": {
                        "ttsName": "Rock music App tts name EN"
                    }
                ]
```

Also the processing of updated structure was checked on current develop and sdl issue was created( sdl does not process such structure as valid ).

dboltovskyi, 2021-09-23:
@GetmanetsIrina @jacobkeeler Looks like currently Core assumes array in `languages` parameter.
In: https://github.com/smartdevicelink/sdl_core/blob/f43fbe015e5c7f84d1bcc1c154aeec6afad217d9/src/components/application_manager/src/application_manager_impl.cc#L2776
And in: https://github.com/smartdevicelink/sdl_core/blob/f43fbe015e5c7f84d1bcc1c154aeec6afad217d9/src/components/application_manager/src/application_manager_impl.cc#L2785

Additionally script [977_SDL_sends_appName_in_vrSynonyms...](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/8_0/977_SDL_sends_appName_in_vrSynonyms_and_ttsName_in_case_of_lower_and_upper_bound_values_of_params_in_json_file.lua) was included in upcoming release `8.0` and it works with array in json files.

So it seems suggested update in https://github.com/smartdevicelink/sdl_atf_test_scripts/issues/2041 is not relevant and #2556 can be closed, and this issue re-opened.

However mentioned json file [JSONWithTtsNameVrSynonymsInSeparatedStructs.json](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/files/jsons/JSON_Language_parameter/JSONWithTtsNameVrSynonymsInSeparatedStructs.json)  requires an update: `appId` for iOS should differ to Android. Otherwise Core reports an error:
```
QUERY_APPS_VALIDATION_FAILED: An Object ID is not unigue
```

theresalech, 2021-11-10:
@dboltovskyi can you please advise if Luxoft is planning to submit a PR to resolve this re-opened issue? Thank you!

dboltovskyi, 2021-11-10:
@theresalech Yes, Luxoft team is planning to submit a fix for this issue.

jacobkeeler, 2022-01-11:
Closed with the merge of #3827

closed 2022-01-11 by jacobkeeler