smartdevicelink/sdl_core #995: [SDL4.0] SDL sends OnSystemRequest (QUERY_APPS) to the app after unsuccessful attempt

KhrystynaDubovyk, 2016-11-07:
**Occurrence:** 
Always

**Preconditions:** 
SDL4.0 feature is enabled in .ini file, SDL and HMI are started.

**Steps:**
1. Register app via 4th protocol.
2. App sends incorrect json on SystemRequest and receives error code in response
3. Bring app to background and then to foreground again or register new app via 4th protocol.

**Expected result:**
SDL does not send OnSystemRequest(QUERY_APPS) to the same app after bringing it to foreground and also to new registered app after unsuccessful attempt to send query json.

**Actual result:**
SDLsends OnSystemRequest(QUERY_APPS) to the same app after bringing it to foreground and also to new registered app after unsuccessful attempt to send query json.

**Note:** To reproduce script using ATF extract files_for scrip_execution_OnSystemRequest_quesry_apps_after_unsuccessful_attempt.tar.gz to ATF_build folder and execute script OnSystemrequest_query_apps_after_unsuccessful_attempt.lua.

**Environment:**
Release: 4.2.0 (master marge)
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
ATF: 962e1eff8664d97af5e59ec4e22f39b9ea76f5eb

**Attachments:**

vvvakulenko, 2017-03-20:
**Reproducible**
SDL: HEAD of https://github.com/smartdevicelink/sdl_core/tree/feature/extended_policy_functionality
(commit d5f63b4f66d9ab945d452dfb1c07bf7ee3ca781e)
ATF: HEAD of https://github.com/smartdevicelink/sdl_atf/tree/develop
(commit aa5beae6ab1726a9b43ed2fa15f66c20d7f7883f)

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

GetmanetsIrina, 2018-06-25:
Defect is reproduced.
SDL branch: develop(2ec732c56e3f06b7043cf02cf98109720d6ff38a)
ATF: develop
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1929

Ypostolov, 2018-07-11:
@GetmanetsIrina 
Please update script for reproducing this issue (SLD considers invalid json from test script as a valid one).

Ypostolov, 2018-10-29:
> @GetmanetsIrina
> Please update script for reproducing this issue (SLD considers invalid json from test script as a valid one).

Comment above not actual.

LitvinenkoIra, 2019-12-11:
Defect is reproduced.
SDL branch: develop(e137556)
ATF: develop
Script: smartdevicelink/sdl_atf_test_scripts#1929

LitvinenkoIra, 2019-12-11:
#2335 - rebased to the latest develop
Fix is partially valid

LitvinenkoIra, 2019-12-11:
There was an atf test script issue,
After https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1929/commits/3d0e1748ee3b99064a92a3b3c5c02819a99f64cb - script passed successfully, so the **fix from #2335 - is valid**

LitvinenkoIra, 2019-12-11:
Full regression successfully passed
SDL: fix_sdl_sends_query_apps_to_t 15253302
Scripts: develop 6f43bfd
ATF: develop 37da7aba

LitvinenkoIra, 2019-12-11:
@theresalech Please triage this issue

theresalech, 2019-12-11:
@LitvinenkoIra We are currently working on finalizing the roadmap with the Steering Committee, so will keep this PR in consideration for upcoming Core releases.

iCollin, 2020-01-13:
Closed via #2335

PR #2335 (2020-01-13): Fix sdl sends OnSystemRequest (QUERY_APPS) to the app
Fixes #995

This PR is ready for review.

### Risk
This PR makes no API changes.

### Plan
Provide ATF tests.

### Summary
In the current implementation SDL does not send OnSystemRequest(QUERY_APPS) to the same app after bringing it to foreground and also to new registered app after unsuccessful attempt to send query json.

### CLA
- [x] I have signed [the CLA](https://docs.google.com/forms/d/e/1FAIpQLSdsgJY33VByaX482zHzi-xUm49JNnmuJOyAM6uegPQ2LXYVfA/viewform)
closed 2020-01-13 by iCollin