smartdevicelink/sdl_core #997: SDL doesn't set unsuccessful "message" value to "info" param in case HMI responds via single UI.RPC when <Interface>.IsReady missing

DrachenkoAnastasiia, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. SDL and HMI are started
2. HMI is initialized and didn't respond on UI.IsReady:
SDL -> HMI: UI.IsReady
HMI -> SDL: HMI didn't repond to UI.IsReady 

**Steps to reproduce:**
1. App is registered
2. App sends to SDL single UI.<RPC> => HMI responds with <resultCode>.

Example with UI.AddSubMenu:

    TC_AddSubMenu_Only_GENERIC_ERROR_Case_1_IsReady_HMI_Does_Not_Respond :
    MOB->SDL :
    Unknown macro: {"position"}

    SDL->HMI : {"id":35,"jsonrpc":"2.0","method":"UI.AddSubMenu","params":{"appID":349706246,"menuID":1000,"menuParams":
    Unknown macro: {"menuName"}

    }}

    HMI->SDL : {"id":35,"jsonrpc":"2.0","result":{"method":"UI.AddSubMenu","message":"error message","code":22}}
    SDL->MOB:
    Unknown macro: {"success"} 

**Actual result:**
1. SDL doesn't send "message" value in "info" param to App if any erroneous(success:false) resultCode with error message received from HMI
2. Issue is reproducible for all single RPCs.
3. Issue is reproducible for all Interfaces: UI, VehicleInfo, TTS, VR

**Expected result:**
In case HMI respond via RPC with "message" param, SDL must retrieve the value of "message" param, set this value to "info" parameter and transfer "info" parameter via corresponding RPC to mobile app:
SDL->MOB:
{"success":false,"resultCode":"GENERIC_ERROR", "info": "error message"} 

**Note:** Issue could be reproduced by ATF:
1. Put to user_modules in test_run folder of ATF build attached files: ATF_Interface_IsReady_missing_SingleRPC_Template.lua, Interfaces_RPC.lua, isReady.lua
2. In test_scripts folder in sdl_atf_test_scripts folder of ATF build put ATF_UI_IsReady_missing_SingleRPC_GENERIC_ERROR.lua

**Environment:**
SDL repo: https://github.com/smartdevicelink/sdl_core.git
SDL branch: master
SDL commit: a6c32b4
HMI: https://github.com/LuxoftSDL/sdl_hmi: 01cb5b49883ed6288e34fcf7db24fd0439d08312
OS: Ubuntu 14.04 LTS

**Attachments:**

JackLivio, 2017-08-15:
@DrachenkoAnastasiia I don't believe this is an issue.

1). If HMI never responds to UI.isReady, then all UI messages should be returned to the mobile with the error code "GENERIC_ERROR".
2). Using Develop without the changes included in pr #1639, I am able to receive a message in the returned "info" field with "success":false

DrachenkoAnastasiia, 2017-08-18:
@Itileda, please, take care of this question.

LuxoftAKutsan, 2017-11-01:
@JackLivio PR closed. I tested issue with attached ATF scripts, and scripts pass. Probably fixed indirectly.

LuxoftAKutsan, 2017-11-01:
Issue not reproduced on latest develop (989bb34)

closed 2017-11-01 by LuxoftAKutsan