smartdevicelink/sdl_core #998: SDL returns IGNORED instead of UNSUPPORTED_RESOURCE for UnsubscribeButton

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. In file hmi_capabilities.json "SEARCH" button exists in capabilities.
2. Core, HMI are started.
3. SDL receives Buttons.GetCapabilities response. "SEARCH" button is not in list for supported buttons.

**Steps:**
1. Send RPC UnsubscribeButton with buttonName = "SEARCH".

**Expected result:**
SDL shall return result_code: UNSUPPORTED_RESOURCE, success: false

**Actual result:**
SDL returns result_code: IGNORED, success: false.

**Note:**
Behaviour of SDL for RPC SubscribeButton in case not supported button is correct.

**Environment:**
Release: 4.2.0 (master marge)
Commit: a6c32b45832653d6b851be9b19355fa35f377e07
OS: Ubuntu 14.04 
Transport: WiFi 
App type: navi
App: Luxoft SPT
Device: HTC One 

**Attachments:**

LevchenkoS, 2017-03-09:
Issue should be resolved after merge of PR #1358

LevchenkoS, 2017-03-17:
#1358 Merged

PR #1621 (2017-08-10): Fix UnsubscribeButton wrong result code in case of UNSUPPORTED_RESOURCE
Fixed `UnsubscribeButton` wrong result code in case of `UNSUPPORTED_RESOURCE`.
This checking was already implemented for `SubscribeButton` rpc, but was not implemented for `UnsubscribeButton`.

In this pull request:
- Function `CheckHMICapabilities()`, which responsible for HMI capabilities permissions checking, was moved to `CommandRequestImpl` class to use from both `SubscribeButton/UnsubscribeButton`.
- Added logs and minor changes in `CheckHMICapabilities()` function
- Added checking HMI capabilities for `UnsubscribeBu [... cut]
closed 2017-08-10 by JackLivio