smartdevicelink/sdl_core #999: SDL does not send default vrHelp to HMI if App was registered with vrSynonyms

KhrystynaDubovyk, 2016-11-07:
**Occurrence:**
Always

**Preconditions:**
1. Core, HMI started.
2. App is registered on HMI.
3. App provided vrSynonyms in RegisterAppInterface.

**Steps:**
1. Send SetGlobalProperties with only "menuTitle" parameter

**Expected result:**
SDL should send default vrHelp and vrHelpTitle

**Actual result:**
SDL do not send vrHelp at all.

**Note:** Problem can be reproduced with attached ATF script.

**Environment:**
SDL branch: master
SDL commit: a6c32b4
HMI: HTML5
OS: Ubuntu 14.04 LTS
Transport: WiFi
Mobile device: HTC One
Mobile OS: Android 4.3
Mobile App version: SPT v.2.1.81 Code: 239
Mobile App type (media/non-media/navi/any): any

**Attachments:**

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality on SDL side with standard use-case.

jacobkeeler, 2018-03-08:
Duplicate of #1306

closed 2018-03-08 by jacobkeeler