smartdevicelink/sdl_core #1046: PolicyTableUpdate must be triggered in case no "certificate" exist at "module_config" section of PolicyTable

TMelnyk, 2016-11-16:


DrachenkoAnastasiia, 2016-11-22:
_**Description:**_
SDL should have a cert before it starts the handshake.
_Proposal:_
Mobile sends start service with encryption true.
If SDL doesn't have a cert it kicks off the policy table update.
If the policy table update fails OR there is no cert in the policy table:
If force secure connection is on - NACK
If force secure connection is off - ACK with encryption false
If a cert is received, begin the handshake
It would also help to expedite the process if a policy table request for a cert was performed when a NAVIGATION app registers and there is no cert.

**_Impact:_** 
SDL 
PolicyManager
TestCases 
Requirements

JackLivio, 2017-04-18:
@TMelnyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio