smartdevicelink/sdl_core #1048: [Policies] CCS: ON by the User

TMelnyk, 2016-11-16:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
1. In policies: each "functional grouping" might have two new params:
`“disallowed_by_ccs_entities_on”:`
``` 
[{
                "entityType": <integer>,
                "entityID": <integer>,
      }]
```
`“disallowed_by_ccs_entities_off”:`
``` 
[{
                "entityType": <integer>,
                "entityID": <integer>,
      }]
```

2. User can go to settings on HMI and enable or disable "CCS" for an "entity" (HMI will notify SDL accordingly):

_Information:_
HMI is responsible for matching User's chosen entity with correct "entityType" and "entityID" when providing the CCS status to SDL.

3. If CCS is **"ON"** for "functional grouping" that
a. contains “disallowed_by_ccs_entities_**on**” with the corresponding "entityType" and "entityID" => such grouping is **USER_DISALLOWED** for all apps that have it assigned
b. contains “disallowed_by_ccs_entities_**off**” with the corresponding "entityType" and "entityID" => such grouping is **USER-ALLOWED** (= processed)
c. **omits** all of above two => such grouping is **USER-ALLOWED**

4. During CCS "ON", the user can go to app's settings and:
a. enable the "functional grouping" particularly for this app (HMI will notify SDL accordingly) => such grouping is **processed** by SDL.
b. disable the "functional grouping" particularly for this app (HMI will notify SDL accordingly) => such grouping will be **USER_DISALLOWED**.

5. After the **CCS** that affects app's grouping(s) **changes**, SDL sends **OnPermissionsChange** (with "user_disallowed" for RPCs from this grouping(s)) to such app.

6. SDL stores "CCS" status between ignition cycles.

_**Impact:**_
SDL 
HTML5 HMI 
HMI_API
TestCases 
Requirements

JackLivio, 2017-04-18:
@TMelnyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio