smartdevicelink/sdl_core #1049: [Policies], [Validation] New CCS-related params in PTU, PreloadedPT, SnapshotPT

TMelnyk, 2016-11-16:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
**1.** New optional
 `“disallowed_by_ccs_entities_on”:`
 ```
[{
                "entityType": <int>,
                "entityID": <int>,
      }]
```
is added to "\<functional grouping\>" ("functional_groupings" section) - of all PTUpdate, SnapshotPT, PreloadedPT.
-> PTUpdate, SnapshotPT, PreloadedPT are valid in both cases **with** and **without** this parameter.
-> PTUpdate, PreloadedPT are invalid in case with **invalid type** of this parameter.

**3.** New optional
`"ccs_consent_groups":`
```
[
                           "<functional grouping>": true/false
           ],
```
is added to "device_data" -> "\<device identifier\>" -> "user_consent_records" -> "\<app id\>" section of SnapshotPT.
_Example:_
```
{
  "policy_table": {
    "device_data": {
      "<deviceID>": {
         "user_consent_records": {
            "<appID>": {
              "consent_groups": {
                "Location": true
             },
            "ccs_consent_groups": [
                           "Location": false
            ],
         "input": "GUI",
         "time_stamp": "2015-10-09T18:07:21Z"
          }
```
-> SnapshotPT are valid in both cases **with** and **without** this parameter.
-> PTUpdate, PreloadedPT are valid only **without** this parameter.

_**Impact:**_
SDL (PolicyManager)
TestCases 
Requirements

JackLivio, 2017-04-18:
@TMelnyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio