smartdevicelink/sdl_core #1050: [Policies] CCS: OFF by the User

TMelnyk, 2016-11-16:


DrachenkoAnastasiia, 2016-11-21:
_**Description:**_
1. In policies: each "functional grouping" might have two new params:
`“disallowed_by_ccs_entities_on”:` 
```
[{
                "entityType": <integer>,
                "entityID": <integer>,
      }]
```
`“disallowed_by_ccs_entities_off”:` 
```
[{
                "entityType": <integer>,
                "entityID": <integer>,
      }]
```
2. User can go to settings on HMI and enable or disable "CCS" for an "entity" (HMI will notify SDL accordingly):
_Information:_
HMI is responsible for matching User's chosen entity with correct "entityType" and "entityID" when providing the CCS status to SDL.
3. If CCS is **"OFF"** for app's "functional grouping" that:
a. contains “disallowed_by_ccs_entities_**on**” with the corresponding "entityType" and "entityID" => such grouping is **allowed**
b. contains “disallowed_by_ccs_entities_**off**” with the corresponding "entityType" and "entityID" => such grouping is **USER_DISALLOWED** for all apps that have it assigned
c. **omits** all of above two => RPCs of such grouping must be processed in terms of other policies settings.
4. After the **CCS** that affects app's grouping(s) **changes,** SDL must update with status "true" "consent_groups" and "ccs_consent_groups" of all impacted apps
SDL sends **OnPermissionsChange** (enabling "functional grouping") to such app.
5. If CCS is **"OFF"** and SDL received OnAppPermissionConsent notification and after that a new app has registered, such app will get the same CCS allowence as other previously connected apps.
6. If SDL receives CCS **"OFF"** with "entityType" and "entityID" and app's "functional grouping" contains “disallowed_by_ccs_entities_**off**” with the corresponding "entityType" and "entityID" pair => the whole grouping is **USER_DISALLOWED** for this app
7. SDL stores "CCS" status between ignition cycles.

**_Impact:_** 
SDL 
HTML5 HMI 
TestCases 
Requirements

JackLivio, 2017-04-18:
@TMelnyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio