smartdevicelink/sdl_core #1057: [SetGlobalProperties] SDL must support new "autoCompleteList" parameter at "KeyboardProperties" struct

mghiumiusliu, 2016-11-22:
*_Description:*_
SDL must support and process new "autoCompleteList" param at SetGlobalProperties RPC

*_Information:*_ 
1. SDL must respond with _"GENERIC_ERROR, success:false"_ in case HMI *_does NOT respond_* during <DefaultTimeout>

_*Notes for HMI expected behavior* (required for Guidelines)_
{quote}
# "The system shall have the ability to receive and store multiple strings for autoCompleteText per app."
# "When the system receives a new list of strings for autoCompleteText for a particular app, the system shall delete the previous list and replace it with the new list for that app."
# "When any of the keyboard layouts are being used, the system shall reference the list of autoCompleteText strings for that app."
# "As the user enters data on the keyboard, the system shall display the autoCompleteText strings which match the entry."
# "The number of matching autoCompleteText strings displayed shall only be limited by the character length constraints of the hmi."
# "The system shall provide the user the ability to select one of the displayed matching autoCompleteText strings without having to enter the entire string."
# "When the user selects one of the displayed matching autoCompleteText string, the system shall submit that entry and not require further user input for submission."
{quote}

. 
*_Impact:*_ 
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

JackLivio, 2017-04-18:
@mghiumiusliu Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio

LevchenkoS, 2017-05-11:
PR #1508