smartdevicelink/sdl_core #1058: [PerformAudioPassThru] SDL must support new "audioPassThruIcon" parameter

KhrystynaDubovyk, 2016-11-22:
**Description**
 SDL must support and process new "audioPassThruIcon" for PerformAudioPassThru RPC

**Information** 
 1. In case HMI does NOT respond during <DefaultTimeout> -> SDL must return GENERIC_ERROR 
2. SDL must validate received response from HMI -> APPLINK-15494, APPLINK-14765
3. Please see the Image struct:

```
<struct name="Image">
  <param name="value" maxlength="65535" type="String" mandatory="true">
    <description>The path to the dynamic image stored on HU or the static binary image itself.</description>
  </param>
  <param name="imageType" type="Common.ImageType" mandatory="true">
    <description>Describes, whether it is a static or dynamic image.</description>
  </param>
</struct>
```

**Notes for HMI expected behavior (required for Guidelines)** 

> 1."The system shall accept requests for AudioPassThru with a reference image."
> 2."When the system receives a request for AudioPassThru with a reference to a valid image, the system shall display the image on the pop-up of the PerformAudioPassThru."
> 3."The system shall accept images of type STATIC and DYNAMIC for AudioPassThru."
> 4."If the image referenced in a request for AudioPassThru is not valid or not available or no image is referenced, the system shall continue processing the request by showing the default image.The system shall provide warning information to the application that the image was not displayed.
>  (Refer H67D for default image)"
> 

**Impact** 
 SDL 
 HTML5 HMI 
 SPT 
 TestCases 
 Requirements 
 Requirements

JackLivio, 2017-04-18:
@KhrystynaDubovyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio

LevchenkoS, 2017-05-11:
PR #1505