smartdevicelink/sdl_core #1059: [AddSubMenu] SDL must support new "subMenuIcon" parameter

KhrystynaDubovyk, 2016-11-22:
**Description**
 SDL must support and correctly handle new param "subMenuIcon" at AddSubMenu RPC 

**Information** 
1. In case HMI does NOT respond during <DefaultTimeout> -> SDL must return GENERIC_ERROR 
2. SDL must validate received response from HMI -> APPLINK-15494, APPLINK-14765
3. All AddSubMenu-related req-s should be applicable
4. Please see the Image struct:
```
<struct name="Image">
  <param name="value" maxlength="65535" type="String" mandatory="true">
    <description>The path to the dynamic image stored on HU or the static binary image itself.</description>
  </param>
  <param name="imageType" type="Common.ImageType" mandatory="true">
    <description>Describes, whether it is a static or dynamic image.</description>
  </param>
</struct>
```

**Notes for HMI expected behavior: (required for Guidelines)**

> 1."The system shall have the ability to display images on sub menus if provided by the application."
> 2."If the application does not provide an image for display on the sub menu, then the system shall not display any image."
> 3."If the image referenced by the application for display on the submenu is invalid or is not available, then the system shall not display any image.
>  The system shall provide a warning information to the application that the SubMenu was added but no image was displayed."
> 4."The system shall scale the image to ensure it fits properly in the space allocated for the display of the image. If the image does not fit properly even after scaling, the system shall not display any image on the submenu."
>  "[Note: Changes to Mobile_Api.xml attached below. Search for changes between ""RMK START --->"" and ""RMK END --->>""
> 2. To remove an icon already sent, the app would have to delete the submenu and add it again without the icon. Otherwise, if the submenu is not deleted, another request with the same submenu id will be rejected.]"
> 

**Impact**
 SDL 
 HTML5 HMI 
 SPT 
 TestCases 
 Requirements

JackLivio, 2017-04-18:
@KhrystynaDubovyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio

LevchenkoS, 2017-05-11:
PR #1490