smartdevicelink/sdl_core #1060: [SetAudioStreamingIndicator] Conditions for SDL to handle related cases

mghiumiusliu, 2016-11-22:
**Description**
SDL must support and process per the similar logic new _SetAudioStreamingIndicator_ RPC

**Information**
1. SDL must validate responses from HMI
2. SDL must respond with _"GENERIC_ERROR, success:false"_ in case HMI *_does NOT respond_* during <DefaultTimeout>

**Noted for HMI expected behavior: (required for Guidelines)**

1. "The Play/Pause button of the media app shall be able to show one of the following icons:
- Play/Pause (currently used icon)
- Play (a play only icon)
- Pause (a pause only icon)
- Buffering (an icon that indicates that audio playback is buffering)"
2. "The icon of the Play/Pause button shall be modifiable by the media app by sending a RPC Request called ""SetAudioPlayingState""."
3. "The RPC Request ""SetAudioPlayingState"" must have a parameter called ""audioPlayingState"". This parameter must be of the enum ""AudioPlayingState""."
4. "The enum ""AudioPlayingState"" must have the following values:
- PLAY_PAUSE (to set the button to the play/pause icon)
- PLAY (to set the button to the play icon)
- PAUSE (to set the button to the pause icon)
- BUFFERING (to set the button to the buffering icon)"
5. "The RPC Response ""SetAudioPlayingStateResponse"" must return:
- SUCCESS when the Play/Pause button is set to the desired icon
- IGNORED when the Play/Pause button was already set to the desired icon
- DISALLOWED when the RPC is used in an invalid HMI level."
6. "The RPC ""SetAudioPlayingState"" shall be 
- ALLOWED within the HMI levels: FULL, LIMITED, BACKGROUND
- DISALLOWED within the HMI levels: NONE"
7. "The RPC ""SetAudioPlayingState"" shall be added to the ""base-x"" (currently it is ""base-4"") functional group."
8. "The default icon for the Play/Pause button shall be the Play/Pause icon (PLAY_PAUSE). "
9. "The default icon must be set to the button when 
- the media app is newly registered on the head unit (after RegisterAppInterface).
- the media app was closed by the user (App enters HMI_NONE)."

**Impact**
SDL 
HTML5 HMI 
SPT 
TestCases 
Requirements

JackLivio, 2017-04-18:
@mghiumiusliu Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio

LevchenkoS, 2017-05-11:
PR #1511