smartdevicelink/sdl_core #1062: [SetMediaClockTimer] SDL must support new parameter "enableSeek" and transfer OnSeekMediaClockTimer notification from HMI to mobile app

KhrystynaDubovyk, 2016-11-22:
**Description**
 SDL must support and process new "enableSeek" param at SetMediaClockTimer
 In case "enableSeek" is true -> SDL expects OnSeekMediaClockTimer from HMI and trabsfer this notification to mobile app

**Information** 
 1. SDL must transfer OnSeekMediaClockTimer notification from HMI to mobile app independently on value of "enableSeek" (true or false) parameter at SetMediaClockTimer_request
2. SDL must validate notifications from HMI 
3. SDL must validate requests from mobile app -> INVALID_DATA
4. SDL must validate received response from HMI

**Notes for HMI requirements (required for Guidelines)**

> 1."A notification will be provided if the user touches (or seeks) to a particular position on the HMI mediaclock timer. The notification will contain a starttime with the position the user has seeked to."

**Impact**
 SDL 
 HTML5 HMI 
 SPT 
 TestCases 
 Requirements

JackLivio, 2017-04-18:
@KhrystynaDubovyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio

LevchenkoS, 2017-05-11:
PR #1492