smartdevicelink/sdl_core #1063: [INI file] Conditions for SDL to create and use 'AppIconsFolder' storage

mghiumiusliu, 2016-11-22:
**Description**
Rules for SDL how to create and use 'AppIconsFolder' storage

**Illustrations**
a. "AppIconsFolder"
|#|Existence |Read/write permissions | SDL behavior|
|1.| no| yes| create folder and use it as assigned|
|2.| yes|yes| use the folder as assigned|
|3.| no|no| log error internally and continue working|
|4.| yes|no| log error internally and continue working|

b. "AppIconsFolderMaxSize"
|#|Default value |Value at .ini file | SDL behavior|
|1.| 1048576 bytes| less than default| use the default value = 1048576 bytes|
|2.| 1048576 bytes|greater that default| use the greater value from .ini file|

**Impact**
SDL 
TestCases 
Requirements

JackLivio, 2017-04-18:
@mghiumiusliu Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to https://github.com/smartdevicelink/sdl_requirements

closed 2017-04-18 by JackLivio

okozlovlux, 2017-05-11:
PR #1506