smartdevicelink/sdl_core #1064: [RegisterAppInterface] SDL must provide the value of "steeringWheelLocation" parameter at response

KhrystynaDubovyk, 2016-11-22:
**Description**
SDL must return the value of "steeringWheelLocation" received from HMI or retrieved from 'HMI_capabilities.json' file

**Information**
SDL must validate received response from HMI 

**Notes for HMI expected behavior (required for Guidelines)**

> 1."The system shall provide the vehicle driver location as part of the response to a registration request.
>  The system shall provide a value of:
>  0 - for Left Hand Drive
>  1 - for Right Hand Drive"
> 2.The system shall source the vehicle driver location information from DID 0xDE00, Byte 4, Bit 6.
> 3.For Vehicle Driver Location information, a value of:
>  0 - means the vehicle is Left Hand Drive
>  1 - means the vehicle is Right Hand Drive
> [Note: Changes to Mobile_Api.xml attached below. Search for changes between ""RMK START--->"" and ""RMK END--->>""]"
> 
> 

**Impact**
 SDL 
 HTML5 HMI 
 SPT (for testing purposes)
 TestCases 
 Requirements

JackLivio, 2017-04-18:
@KhrystynaDubovyk Closing since this is a requirement and not an issue. If this CRQ is still being worked on please relocate to [https://github.com/smartdevicelink/sdl_requirements](url)

closed 2017-04-18 by JackLivio

okozlovlux, 2017-05-11:
PR #1509