smartdevicelink/sdl_core #1096: Perform Interaction Manual Mode Still Requires Vr Response

JackLivio, 2016-12-07:
### Description

Core does not send Perform Interaction Response to Mobile when using manual mode (i.e. only using ui.performinteraction). Core will not send the response to mobile unless it receives a response for both ui and vr perform interaction. 

### Information
This image shows the implemented flow for manual only. Why does the HMI HAVE to respond to VR.PerformInteraction in manual only mode?
[image]

### Impact
Modules that do not have VR implemented and will not be able to respond to VR.PerformInteraction.

jacobkeeler, 2018-08-29:
Seems that the issue here is that TTS information is included in the `VR.PerformInteraction` request, so I believe that there would need to be a change of the design of these RPCs in order to fix this issue.

chinlin0924, 2019-04-19:
Hi, when will this be fixed?

jacobkeeler, 2022-03-14:
Closing, because a proposal would be needed to change this behavior at this point

closed 2022-03-14 by jacobkeeler