smartdevicelink/sdl_core #1105: Segmentation fault when run smartDeviceLinkCore

grimlockrocks, 2016-12-10:
Hello, 

I am very new to SDL. Just followed the instruction to build everything, after many modifications here and there, the build was successful, but when I ran 
LD_LIBRARY_PATH=. ./smartDeviceLinkCore

I got 
zsh: segmentation fault (core dumped)  LD_LIBRARY_PATH=. ./smartDeviceLinkCore

GDB did not provide any useful information:
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff5ddb4b6 in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() () from /usr/lib64/libstdc++.so.6

I am running it on Amazon Linux Bare Metal, a version of Red Hat Linux I guess. 

Any help is appreciated. Thank you!

vlantonov, 2016-12-12:
@grimlockrocks Please try this sequence:
LD_LIBRARY_PATH=$LD_LIBRARY_PATH:.
./smartDeviceLinkCore

grimlockrocks, 2016-12-12:
Thanks @vlantonov, but it did not seem to work:

% export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:. 
% ./smartDeviceLinkCore 
zsh: segmentation fault (core dumped)  ./smartDeviceLinkCore
% echo $LD_LIBRARY_PATH 
:.

vlantonov, 2016-12-13:
@grimlockrocks If the smartDeviceLinkCore was tarted, there should be smartDeviceLinkCore.log which is expected to contain more information about the calls. Please try to overwrite inside bin folder the smartDeviceLinkCore with one from src/appMain/

grimlockrocks, 2016-12-13:
Thank you. There is no log file generated, my guess is that even though the program ran, it did not even get started with the first phase. I tried to run under src/appMain or point out LD_LIBRARY_PATH to that directory, same result. 

It is not a big deal, but I really hope SDL_Core can be running on Red Hat Enterprise Linux (RHEL).

vlantonov, 2016-12-14:
@grimlockrocks In general this behavior is not expected. If you are interested in solving it please give additional details like - the branch and commit, the compilation configuration, call trace.

JackLivio, 2017-01-11:
@grimlockrocks Can you try using GDB and step through [main()](https://github.com/smartdevicelink/sdl_core/blob/master/src/appMain/main.cc#L118) at the start of the application to see where the crash happens?

JackLivio, 2018-03-08:
Closing due to inactivity. Please post updated issue if still relevant.

closed 2018-03-08 by JackLivio