smartdevicelink/sdl_core #1110: SDL core crashes when BT Disconnect/Connect multiple times

madhuy, 2016-12-14:
it crashes 

https://github.com/smartdevicelink/sdl_core/blob/master/src/3rd_party-static/MessageBroker/src/server/mb_tcpserver.cpp#L104

we tried with null checking 
as below 

if ((buf.empty()) || (&buf[0] == nullptr)) {
      return false;
    }

still it crashes. Could you please invetigate.

madhuy, 2016-12-14:
As you can see from backtrace_835.txt, the crash happened at 0x4883b326.
According to smaps_835.txt, libSmartDeviceLinkCore.so is loaded from 0x47370000.
So the problem happens in libSmartDeviceLinkCore.so at the offset of 0x014cb326.
Based on the debug infomation included in libSmartDeviceLinkCore.so, the address is corresponding to the line 104 in mb_tcpserver.cpp:
[smaps_835.txt](https://github.com/smartdevicelink/sdl_core/files/651825/smaps_835.txt)
[backtrace_835.txt](https://github.com/smartdevicelink/sdl_core/files/651826/backtrace_835.txt)

jnuie, 2016-12-14:
The debug information is the following:

`$ readelf --debug-dump=decodedline libSmartDeviceLinkCore.so | grep '0x14cb[234][0-9a-f][0-9a-f]'`
`mb_tcpserver.cpp                              85           0x14cb20a`
`mb_tcpserver.cpp                              88           0x14cb21a`
`mb_tcpserver.cpp                              89           0x14cb222`
`mb_tcpserver.cpp                              91           0x14cb232`
`mb_tcpserver.cpp                              91           0x14cb254`
`mb_tcpserver.cpp                              95           0x14cb258`
`mb_tcpserver.cpp                              96           0x14cb260`
`mb_tcpserver.cpp                              96           0x14cb276`
`mb_tcpserver.cpp                              97           0x14cb288`
`mb_tcpserver.cpp                              97           0x14cb29e`
`mb_tcpserver.cpp                              98           0x14cb2aa`
`mb_tcpserver.cpp                              98           0x14cb2d6`
`mb_tcpserver.cpp                              99           0x14cb2f8`
`mb_tcpserver.cpp                             101           0x14cb2fc`
`mb_tcpserver.cpp                             104           0x14cb31e`
`mb_tcpserver.cpp                             104           0x14cb32e`
`mb_tcpserver.cpp                             104           0x14cb340`
`mb_tcpserver.cpp                             104           0x14cb344`
`mb_tcpserver.cpp                             104           0x14cb346`
`mb_tcpserver.cpp                             105           0x14cb34a`
`mb_tcpserver.cpp                             107           0x14cb34e`
`mb_tcpserver.cpp                             107           0x14cb37a`
`mb_tcpserver.cpp                             112           0x14cb39c`
`mb_tcpserver.cpp                             114           0x14cb3b8`
`mb_tcpserver.cpp                             115           0x14cb3c2`
`mb_tcpserver.cpp                             116           0x14cb3cc`
`mb_tcpserver.cpp                             118           0x14cb3e0`
`mb_tcpserver.cpp                             118           0x14cb3e8`
`mb_tcpserver.cpp                             118           0x14cb3f8`
`mb_tcpserver.cpp                             118           0x14cb3fc`
`mb_tcpserver.cpp                             118           0x14cb3fe`

jacobkeeler, 2016-12-16:
Hello @madhuy, could you give me information on your setup? Are you on master or another release? 

I am currently unable to reproduce the issue on my own machine, could you please provide SmartDeviceLinkCore.log file leading up to the crash?

madhuy, 2016-12-16:
@jacobkeeler We are using 4.1 SDL core. attaching SDL core.
[SmartDeviceLinkCore_1107.txt](https://github.com/smartdevicelink/sdl_core/files/658121/SmartDeviceLinkCore_1107.txt)

`DEBUG [31 Dec 2012 23:00:00,749][TransportManager] bytes_sent < 03035624416)
ERROR [31 Dec 2012 23:00:00,749][TransportManager] Send failed for connection 0xb55009f0, error code 0 (Success)
DEBUG [31 Dec 2012 23:00:00,749][SDLMain] SIGSEGV signal has been caught
INFO  [31 Dec 2012 23:00:00,948][HMIMessageHandler] Successfully sent to observer
INFO  [31 Dec 2012 23:00:00,948][HMIMessageHandler] Received message from hmi
INFO  [31 Dec 2012 23:00:00,948][HMIMessageHandler] Message from hmi given away.`

jacobkeeler, 2016-12-20:
Hello @madhuy, while the log file you provided was helpful, I think that the actual log file generated by Core (which includes filenames, line numbers, and trace logs) is needed to pinpoint the problem. Could you please post the SmartDeviceLinkCore.log file that was created in your bin directory?

Also, what OS/version are you running SDL Core on?

jnuie, 2016-12-21:
@jacobkeeler
Here it is.
[SmartDeviceLinkCore_835.log.zip](https://github.com/smartdevicelink/sdl_core/files/665943/SmartDeviceLinkCore_835.log.zip)

jacobkeeler, 2016-12-21:
@jnuie This looks like the log file for #1113, I don't see the crash in this log. 

In addition, the log you provided does not include file/function names and line numbers (which is useful for following trace logs). This information should be included by default in the generated log file unless you changed the format in your log4cxx.properties. Have you modified this file?

The default appender settings for SmartDeviceLinkCore.log should look like this
```
# Log file for all SmartDeviceLinkCore messages
log4j.appender.SmartDeviceLinkCoreLogFile=SafeFileAppender
log4j.appender.SmartDeviceLinkCoreLogFile.File=SmartDeviceLinkCore.log
log4j.appender.SmartDeviceLinkCoreLogFile.append=true
log4j.appender.SmartDeviceLinkCoreLogFile.DatePattern='.' yyyy-MM-dd HH-mm
log4j.appender.SmartDeviceLinkCoreLogFile.ImmediateFlush=true
log4j.appender.SmartDeviceLinkCoreLogFile.layout=org.apache.log4j.PatternLayout
log4j.appender.SmartDeviceLinkCoreLogFile.layout.ConversionPattern=%-5p [%d{dd MMM yyyy HH:mm:ss,SSS}][%t][%c] %F:%L %M: %m%n
```

jacobkeeler, 2016-12-21:
@madhuy @jnuie Another thing that might help is turning on debug messages for the message broker by uncommenting [this line](https://github.com/smartdevicelink/sdl_core/blob/master/src/3rd_party-static/MessageBroker/include/MBDebugHelper.h#L16) in the message broker, and posting the logs you receive from that.

```
#define DEBUG_ON
```

madhuy, 2016-12-21:
@jacobkeeler 
Plese check log SmartDeviceLinkCore_1107.txt look for  "SIGSEGV signal has been caught" 
@jnuie Please enable DEBUG_ON  and repro that issue.

jnuie, 2016-12-22:
The attached archive includes the latest test result.
[811_2_1.zip](https://github.com/smartdevicelink/sdl_core/files/669380/811_2_1.zip)

madhuy, 2016-12-22:
@jacobkeeler  The log pattern we used is,
`# Log file for all SmartDeviceLinkCore messages
log4j.appender.SmartDeviceLinkCoreLogFile=SafeFileAppender
log4j.appender.SmartDeviceLinkCoreLogFile.File=SmartDeviceLinkCore.log
log4j.appender.SmartDeviceLinkCoreLogFile.append=true
log4j.appender.SmartDeviceLinkCoreLogFile.DatePattern='.' yyyy-MM-dd HH-mm
log4j.appender.SmartDeviceLinkCoreLogFile.ImmediateFlush=true
log4j.appender.SmartDeviceLinkCoreLogFile.layout=org.apache.log4j.PatternLayout
log4j.appender.SmartDeviceLinkCoreLogFile.layout.ConversionPattern=%-5p [%d{dd MMM yyyy HH:mm:ss,SSS}][%t][%c] %F:%L %M: %m%n`

madhuy, 2017-01-12:
@jacobkeeler attaching SmartDeviceLinkcore.log with message broker logs enabled

[SmartDeviceLinkCore.txt.zip](https://github.com/smartdevicelink/sdl_core/files/700393/SmartDeviceLinkCore.txt.zip)

madhuy, 2017-01-12:
crashed 
`DEBUG [11 Jan 2017 16:48:10,940][0xb5bff3e0][ApplicationManager] /data/sdl_core/src/components/application_manager/src/application_manager_impl.cc:188 ApplicationSharedPtr application_manager::FindApp: Unable to find application
TRACE [11 Jan 2017 16:48:10,940][0xb5bff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/hmi/on_app_unregistered_notification.cc:48 commands::OnAppUnregisteredNotification::Run: Enter
TRACE [11 Jan 2017 16:48:10,940][0xb5bff3e0][ApplicationManager] /data/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1600 ApplicationManagerImpl::SendMessageToHMI: Enter
INFO  [11 Jan 2017 16:48:10,940][0xb5bff3e0][ApplicationManager] /data/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1624 ApplicationManagerImpl::SendMessageToHMI: Attached schema to message, result if valid: 1
TRACE [11 Jan 2017 16:48:10,940][0xb5bff3e0][ApplicationManager] /data/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1907 ApplicationManagerImpl::ConvertSOtoMessage: Enter
DEBUG [11 Jan 2017 16:48:10,940][0xb5bff3e0][ApplicationManager] /data/sdl_core/src/components/application_manager/src/application_manager_impl.cc:1919 ApplicationManagerImpl::ConvertSOtoMessage: Message with protocol: 1
DEBUG [11 Jan 2017 16:48:10,941][0xb39ff3e0][SDLMain] /data/sdl_core/src/appMain/life_cycle.cc:316 {anonymous}::sig_handler: SIGSEGV signal has been caught`

jacobkeeler, 2017-01-12:
The thread on which the crash occurs is listed as `0xb39ff3e0`.

Looking at the last message for thread `0xb39ff3e0` before the crash:
```
INFO  [11 Jan 2017 16:48:10,933][0xb39ff3e0][TransportManager] /data/sdl_core/src/components/transport_manager/src/uie/threaded_connection.cc:177 transport_adapter::ThreadedConnection::WriteThreadDelegate::threadMain: removing message (#3013604320) 
```

This message points to a file which is not contained in the SDL core repository (.../src/uie/threaded_connection.cc), so it appears to me that the crash is happening due to issues in your local copy of SDL, not due to an issue in the main project.

madhuy, 2017-01-13:
@jacobkeeler  We found the root cause and it is mainly timing issue. some times our porting layer calling controller()->DataSendFailed() when the connection is destroyed. it is causing the crash . We fixed in our porting layer.  It is not reproducible now. we are testing now the fix. I will close this ticket once testing done.

madhuy, 2017-01-18:
It is fixed. Closing now

closed 2017-01-18 by madhuy