smartdevicelink/sdl_core #1113: SDL core sending onAppUnregistered to HMI when first appregistered with SDL core

madhuy, 2016-12-16:
[SmartDeviceLinkCore_onAppUnregistered.txt](https://github.com/smartdevicelink/sdl_core/files/656196/SmartDeviceLinkCore_onAppUnregistered.txt)

Based on the logs it happens only for the first app which register after starting SDL core
`INFO  [15 Dec 2016 18:45:20,847][ApplicationManager] Unregistering application with app_id 65537 because of HMI language(s) mismatch.`
after onApregistererd
1. it calls [CheckApplication](https://github.com/smartdevicelink/sdl_core/blob/aff7f09a4ec07becfb64dc78992580061d3342ff/src/components/application_manager/src/hmi_language_handler.cc#L138)
2. this value is_need_handle_wrong_language become true and calls HandleWrongLanguageApp() method. 

I am not sure is it because of timing or some thing else.

madhuy, 2016-12-16:
This happens
1. SDL app running in HU
2. IGN OFF in HU
3. IGN ON in HU.

jacobkeeler, 2016-12-20:
@madhuy In a similar vein to #1110, I think the generated SmartDeviceLinkCore.log in the bin directory would be useful in this case, as I am unable to reproduce the issue on my side. Could you please provide this log?

jacobkeeler, 2016-12-21:
Based on the CheckApplication method, it looks like is_need_handle_wrong_language is only set if CheckApplication is called with the same app ID twice (the first with app.second set to true). 

```cpp
sync_primitives::AutoLock lock(apps_lock_);
Apps::iterator it = apps_.find(app.first);
if (apps_.end() == it) {
  LOG4CXX_INFO(logger_,
               "Adding application id "
                   << app.first
                   << " Application registered: " << app.second);
  apps_.insert(app);
  return;
}
is_need_handle_wrong_language = apps_[app.first];
```

The only two places where CheckApplication is called are in the [on_event](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/hmi_language_handler.cc#L117-L150) function (where the event is OnAppRegistered) and the VerifyWithPersistedLanguages function. In the case of the latter there would be an INFO message if CheckApplication was called...
```cpp
LOG4CXX_INFO(logger_,
             "Some languages gotten from HMI have "
             "mismatch with persisted values.");
```
...and this message does not appear in the log.

This seems to imply that this component is receiving two OnAppRegistered events for the same app_id, although I have not be able to figure out the exact reason yet.

madhuy, 2016-12-21:
@jacobkeeler , if Mobile application registers twice it should return error APPLICATION_REGISTERED_ALREADY from [here](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc#L208)

jacobkeeler, 2016-12-22:
@madhuy This isn't the result of receiving two RegisterAppInterface messages, but instead caused by firing two OnAppRegistered events (both as a result of messages sent FROM Core). 

The obvious place where this would be fired is in the OnAppRegistered class ([and it is](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/hmi/on_app_registered_notification.cc#L50)), but it is also fired from the [OnApplicationRegistered function](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/application_manager_impl.cc#L338) in the ApplicationManager, and this function is called when sending out a [RegisterAppInterface response](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/hmi/on_app_registered_notification.cc#L50).

It seems that this redundant event is what is causing this issue, so removing [this block](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/application_manager_impl.cc#L338-L347) may fix your issue.

It appears that the reason it doesn't happen on my machine is that the appID parameter seems to be missing from one of the events:
```
ERROR [22 Dec 2016 10:58:39,446][0x7f70627fc700][ApplicationManager] /home/jacobkeeler/Documents/sdl_core/src/components/application_manager/src/hmi_language_handler.cc:135 HMILanguageHandler::on_event: Message doesn't contain parameter app_id
```
This means that Core is not calling CheckApplication a second time. 

I do not know yet why it isn't able to find the appID parameter on my end (it appears in the OnAppRegistered message), but that is another issue (possibly with the SDL HMI).

madhuy, 2016-12-28:
@jacobkeeler 
with your suggestion by  commenting [this block ](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/application_manager_impl.cc#L338-L347)  I don't see below message.
`Unregistering application with app_id 65537 because of HMI language(s) mismatch.`

do you think that code is redundant?

jacobkeeler, 2017-01-05:
@madhuy Looking deeper into this, it looks to me like the issue lies in the fact that the same event was used for two different purposes. 

The original event (fired after an OnAppRegistered message is received) is meant to be received by the [SDLActivateAppRequest](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/hmi/sdl_activate_app_request.cc#L172-L177) class, which immediately unsubscribes from that event. This event is then received by the HMI language handler, which also receives the "extra" event sent from the OnApplicationRegistered function. 

It seems to me that the two uses of the event are unrelated, so the best solution would likely be where two different events were used rather than reusing the BasicCommunication_OnAppRegistered event ID as is done currently.

I am not absolutely certain whether just removing the extra event will cause any problems, as it is not entirely clear to me why it was added in the first place, but I have not yet run into any issues after removing the block.

In short, I do not know whether the event is entirely redundant or just being used improperly.

@LuxoftAKutsan or @mghiumiusliu The code mentions a specific JIRA issue (APPLINK-20764) from before everything was moved to GitHub, does this have a corresponding GitHub issue?

jacobkeeler, 2017-01-06:
Another thing that I noticed while looking through the HMI language handler. The CheckApplication logic seems like it is reversed from what it should be:

```
is_need_handle_wrong_language = apps_[app.first];
```

Where apps_[app.first] would only be true if CheckApplication was called first with app.second set to false. The only case where this happens is when the OnAppRegistered event is first received, and CheckApplication is called again later (in this case by the second OnAppRegistered event).

The only case where apps_[app.first] would be false would be where an actual language mismatch occurred (detected in the [VerifyWithPersistedLanguages](https://github.com/smartdevicelink/sdl_core/blob/aff7f09a4ec07becfb64dc78992580061d3342ff/src/components/application_manager/src/hmi_language_handler.cc#L269) function) before the OnAppRegistered event was received, which seems to me when it should "HandleWrongLanguageApp".

While this seems like a more obvious flaw, it should not cause the issue that you are having on it's own, as only one OnAppRegistered event should be sent out, resulting in CheckApplication only being called once. Only combined with the redundant event would this logic cause any issues when there is no language mismatch.

With more information about exactly why this logic was included in the first place, I may be able to determine whether this is an actual mistake or not.

jacobkeeler, 2017-01-17:
@madhuy Since you were able to provide logs with the file name information included for #1110, could you provide the same for this issue? 

It looks like we have a pretty good idea where the issue is, but the more detailed log format could make it clearer what series of events leads to this issue (and would be easier to compare to my own setup).

madhuy, 2017-01-26:
@jacobkeeler sorry for late response. Attaching logs
[SmartDeviceLinkCore.log.zip](https://github.com/smartdevicelink/sdl_core/files/733279/SmartDeviceLinkCore.log.zip)

mghiumiusliu, 2017-01-26:
@jacobkeeler JIRA issue was task: Introduce usage of internal events in SDL. It was resolved February 2016 - there was no need in this issue. Instead we have PR: https://github.com/CustomSDL/sdl_panasonic/pull/168.

With https://github.com/CustomSDL/sdl_panasonic/pull/168 there was introduced workaround to notify HMILanguageHadler about app registration finish.

madhuy, 2017-02-08:
@jacobkeeler Do you have any update on this.

jacobkeeler, 2017-02-13:
@madhuy I do not see the error message or OnAppUnregistered notification in the most recent log that you posted.

jacobkeeler, 2017-02-13:
@madhuy Alright, using this log I can see that the reason is not because of the duplicate event. The appID parameter is set as `"params" : "appID"` in the duplicate event as opposed to `"params" : "application": "appID"` in the actual OnAppRegistered message, this is the reason behind the `Message doesn't contain parameter app_id` message I was getting. 

It appears that the actual reason behind this issue is that, for some reason, Core is sending out 2 RegisterAppInterfaceResponse messages in the `SendRegisterAppInterfaceResponseToMobile` function.

```
TRACE [31 Dec 2012 20:06:21,298][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc:171 commands::RegisterAppInterfaceRequest::Run: Enter
TRACE [31 Dec 2012 20:06:21,374][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc:367 commands::RegisterAppInterfaceRequest::SendRegisterAppInterfaceResponseToMobile: Enter
TRACE [31 Dec 2012 20:06:21,380][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_response.cc:46 commands::RegisterAppInterfaceResponse::Run: Enter
TRACE [31 Dec 2012 20:06:21,805][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_response.cc:46 commands::RegisterAppInterfaceResponse::Run: Exit
TRACE [31 Dec 2012 20:06:21,837][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_response.cc:46 commands::RegisterAppInterfaceResponse::Run: Enter
TRACE [31 Dec 2012 20:06:22,192][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_response.cc:46 commands::RegisterAppInterfaceResponse::Run: Exit
DEBUG [31 Dec 2012 20:06:22,202][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc:616 commands::RegisterAppInterfaceRequest::SendRegisterAppInterfaceResponseToMobile: Application with app_id = 65537 doesn't exist.
TRACE [31 Dec 2012 20:06:22,203][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc:367 commands::RegisterAppInterfaceRequest::SendRegisterAppInterfaceResponseToMobile: Exit
TRACE [31 Dec 2012 20:06:22,223][0xb59ff3e0][Commands] /data/sdl_core/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc:171 commands::RegisterAppInterfaceRequest::Run: Exit
```

I don't see any obvious way that this could happen on my end, are there any conditions in your local copy of `register_app_interface_request.cc` (specifically in the [SendRegisterAppInterfaceResponseToMobile](https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/register_app_interface_request.cc#L637) function) where two responses might be sent out in a row? 

Also, if you can't find this case, could you post your copy of `register_app_interface_request.cc` so that I can make sure this issue isn't caused by local changes?

madhuy, 2017-02-14:
@jacobkeeler Please find the attached local copy of register_app_interface_request.cc
[register_app_interface_request.cc.zip](https://github.com/smartdevicelink/sdl_core/files/772798/register_app_interface_request.cc.zip)

madhuy, 2017-02-14:
@jacobkeeler Yes I do see 2 times sendResponse() . I am not sure if it is merge issue or some thing else. I will check by fixing in our repo.

madhuy, 2017-02-15:
@jacobkeeler Thank you . by removing 2nd sendresponse() fixes issue. Sorry of inconvenience.

closed 2017-02-15 by madhuy