smartdevicelink/sdl_core #1174: Core's DateTime implementation is incorrect

joeljfischer, 2017-01-13:
As of SDL Core v4.2.0, [the current spec of the "DateTime" struct](https://github.com/smartdevicelink/sdl_core/blob/4.2.0/src/components/interfaces/MOBILE_API.xml#L2280) is as follows:

```xml
<struct name="DateTime">
   <param name="second" type="Integer" minvalue="0" maxvalue="60" mandatory="true">
      <description>Seconds part of time</description>
   </param>
   <param name="minute" type="Integer" minvalue="0" maxvalue="59" mandatory="true">
      <description>Minutes part of time</description>
   </param>
   <param name="hour" type="Integer" minvalue="0" maxvalue="23" mandatory="true">
      <description>Hours part of time. Note that this structure accepts time only in 24 Hr format</description>
   </param>
   <param name="day" type="Integer" minvalue="1" maxvalue="31" mandatory="true">
      <description>Day of the month</description>
   </param>
   <param name="month" type="Integer" minvalue="1" maxvalue="12" mandatory="true">
      <description>Month of the year</description>
   </param>
   <param name="year" type="Integer" maxvalue="4095" mandatory="true">
      <description>The year in YYYY format</description>
   </param>
   <param name="tz_hour" type="Integer" minvalue="-12" maxvalue="14" defvalue="0" mandatory="true">
      <description>Time zone offset in Hours wrt UTC.</description>
   </param>
   <param name="tz_minute" type="Integer" minvalue="0" maxvalue="59" defvalue="0" mandatory="true">
      <description>Time zone offset in Min wrt UTC.</description>
   </param>
</struct>
```

However, in [SDL Evolution proposal SDL-0004](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0004-sendlocation-updates.md), the spec should have been implemented as follows:

```xml
<struct name="DateTime">                
    <param name="millisecond" type="Integer" minvalue="0" maxvalue="999" mandatory="false">
       <description>Milliseconds </description>     
    </param>         
    <param name="second" type="Integer" minvalue="0" maxvalue="60" mandatory="false">
    <description>Seconds part of time</description>     
    </param>            
    <param name="minute" type="Integer" minvalue="0" maxvalue="59" mandatory="false">
      <description>Minutes part of time</description>       
    </param>            
    <param name="hour" type="Integer" minvalue="0" maxvalue="23" mandatory="false">
    <description>Hours part of time. Note that this structure accepts time only in 24 Hr format</description>       
    </param>            
  <param name="day" type="Integer" minvalue="1" maxvalue="31" mandatory="false">
    <description>Day of the month</description>     
    </param>            
    <param name="month" type="Integer" minvalue="1" maxvalue="12" mandatory="false">
      <description>Month of the year</description>      
    </param>            
    <param name="year" type="Integer" maxvalue="4095" mandatory="false">            
        <description>The year in YYYY format</description>      
    </param>            
    <param name="tz_hour" type="Integer" minvalue="-12" maxvalue="14" mandatory="false">
       <description>Time zone offset in Hours with regard to UTC.</description>     
    </param>            
    <param name="tz_minute" type="Integer" minvalue="0" maxvalue="999" mandatory="false">
       <description>Time zone offset in Min with regard to UTC. </description>      
    </param>            
</struct>  
```

We will have to correct this with a major version update and Mobile library corner case updates.

LuxoftAKutsan, 2017-01-23:
@joeljfischer DateTime specifying time stamp but not time interval.
Accuracy of  DateTime time stamp is second. 
Making any fields of DateTime optional will make it undefined. 
Any way it is  to raise accuracy to millisecond, But what is use case? Note that accuracy of Unix time stamp is second, and currently DateTime is related to it.

joeljfischer, 2017-01-23:
@LuxoftAKutsan it's irrelevant. The decision was made by the SDL Steering Committee and Core is in conflict.

theresalech, 2017-02-15:
Core is in conflict of [SDL Evolution Proposal 004](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0004-sendlocation-updates.md).  The SDLC has agreed that a hotfix should be issued to resolve.

theresalech, 2017-02-15:
[image] [Core's DateTime implementation is incorrect](https://trello.com/c/0403bw3t/33-core-s-datetime-implementation-is-incorrect)

PR #1316 (2017-02-27): Fix DateTime implementation
By [SDL 0004](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0004-sendlocation-updates.md), all of the DateTime parameters should be made optional and the millisecond field should be added to the struct.

Fixes #1174
closed 2017-02-27 by jacobkeeler