smartdevicelink/sdl_core #1175: Core's GetWayPoints request implementation is incorrect

joeljfischer, 2017-01-13:
As of SDL Core v4.2.0, [the current spec of the GetWayPoints request](https://github.com/smartdevicelink/sdl_core/blob/4.2.0/src/components/interfaces/MOBILE_API.xml#L5273) is as follows:

```xml
<function name="GetWayPoints" functionID="GetWayPointsID" messagetype="request">    
  <description>Request for getting waypoint/destination data.</description>
  <param name="wayPointType" type="WayPointType" defvalue="ALL" mandatory="false">
    <description>To request for either the destination only or for all waypoints including destination</description>
  </param>
</function>
```

However, in [SDL Evolution proposal SDL-0003](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0003-last-mile-navigation.md#additions-to-mobile_api), the spec should have been implemented as follows (note the mandatory-ness of the parameter):

```xml
<function name="GetWayPoints" functionID="GetWayPointsID" messagetype="request">        
    <description>Request for getting waypoint/destination data.</description>   
    <param name="wayPointType" type="WayPointType " mandatory="true">   
<description>To request for either the destination only or for all waypoints including destination</description>
    </param>    
</function>
```

LuxoftAKutsan, 2017-01-23:
Can be done in next releases.

theresalech, 2017-02-15:
Core is in conflict of [SDL Evolution Proposal 0003](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0003-last-mile-navigation.md).  The SDLC has agreed that a hotfix should be issued to resolve.

theresalech, 2017-02-15:
[image] [Core's GetWayPoints request implementation is incorrect](https://trello.com/c/xycxC71x/34-core-s-getwaypoints-request-implementation-is-incorrect)

jacobkeeler, 2017-02-28:
Another, likely more troubling, issue I noticed while fixing this issue is that the `GetWayPoints` response in the Mobile API is missing the `waypoints` field.

Current definition of `GetWayPoints` response in `MOBILE_API.xml`:
```xml
  <function name="GetWayPoints" functionID="GetWayPointsID" messagetype="response">
    <param name="success" type="Boolean" platform="documentation">
      <description> true, if successful; false, if failed </description>
    </param>
    <param name="resultCode" type="Result" platform="documentation">
      <description>See Result</description>
      <element name="SUCCESS"/>
      <element name="INVALID_DATA"/>
      <element name="TIMED_OUT"/>
      <element name="GENERIC_ERROR"/>
      <element name="REJECTED"/>
      <element name="UNSUPPORTED_RESOURCE"/>
      <element name="IGNORED"/>
      <element name="IN_USE"/>
      <element name="DISALLOWED"/>
    </param>
    <param name="info" type="String" maxlength="1000" mandatory="false" platform="documentation">
       <description>Provides additional human readable info regarding the result.</description>
    </param>
  </function>
```

Definition of `GetWayPoints` response according to [SDL-0003](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0003-last-mile-navigation.md):
```xml
<function name="GetWayPoints" functionID="GetWayPointsID" messagetype="response">       
    <param name="success" type="Boolean" platform="documentation">  
        <description> true, if successful; false, if failed </description>
    </param>    
    <param name="resultCode" type="Result" platform="documentation">    
        <description>See Result</description>
        <element name="SUCCESS"/>
        <element name="INVALID_DATA"/>
        <element name="TIMED_OUT"/>
        <element name="GENERIC_ERROR"/>
        <element name="REJECTED"/>
        <element name="UNSUPPORTED_RESOURCE"/>
        <element name="IGNORED"/>
        <element name="IN_USE"/>
        <element name="DISALLOWED"/>
    </param>    
    <param name="info" type="String" maxlength="1000" mandatory="false" platform="documentation">   
        <description>Provides additional human readable info regarding the result.</description>
    </param>    
    <param name="wayPoints" type="LocationDetails" mandatory="false" array="true" minsize="1" maxsize="10"> 
        <description>See LocationDetails</description>
    </param>    
</function>  
```

PR #1339 (2017-04-05): Fix WayPoints API implementation
Fixes #1175
closed 2017-04-05 by jacobkeeler