smartdevicelink/sdl_core #1176: Core's WayPoints API function ids are incorrect

joeljfischer, 2017-01-13:
As of SDL v4.2.0, [the current spec of the WayPoints API request function ids](https://github.com/smartdevicelink/sdl_core/blob/4.2.0/src/components/interfaces/MOBILE_API.xml#L2226) are as follows:

```xml
<element name="GetWayPointsID" value="41" hexvalue="29" />
<element name="SubscribeWayPointsID" value="42" hexvalue="2A"/>
<element name="UnsubscribeWayPointsID" value="43" hexvalue="2B"/>
```

However, in [SDL Evolution proposal SDL-0003](https://github.com/smartdevicelink/sdl_evolution/blob/master/proposals/0003-last-mile-navigation.md#additions-to-mobile_api), the spec should have been implemented as follows:

```xml
<element name="GetWayPointsID" value="45" hexvalue="2D" />
<element name="SubscribeWayPointsID" value="46" hexvalue="2E" />
<element name="UnsubscribeWayPointsID" value="47" hexvalue="2F" />
```

LuxoftAKutsan, 2017-01-23:
@joeljfischer value of RPC are incremental. There are no gaps in it. So, if in SDL Evolution value of GetWayPointsID is 45 it means that there some other RPS's from 41 - 43. Any way changing values of RPC is not separate changes. It should be related to adding or removing some RPC.

joeljfischer, 2017-01-23:
Yes, I understand, they are incremental. It was decided that the Remote Control RPCs would occupy spaces 41-43, but they haven't been merged in yet. Therefore, what is currently in Core conflicts with those APIs. That's why the steering committee made the decision it did.

jacobkeeler, 2017-04-05:
Fixed with #1339

closed 2017-04-05 by jacobkeeler