smartdevicelink/sdl_core #1178: Core sends an onAppInterfaceUnregistered notification when UnregisterAppInterface is sent from Mobile

asm09fsu, 2017-01-13:
It seems as though we have an issue with core sending both an `onAppInterfaceUnregistered` notification and an `UnregisterAppInterface` response when the mobile proxy sends over `UnregisterAppInterface`. 
This should not be the case, as the notification should only be sent if there is an unregistration from the core side that was not caused by the mobile side. The notification also has a reason of -1, which is not a valid value. 

We should remove the notification from being sent when the `UnregisterAppInterface` RPC is sent.

JackLivio, 2017-04-18:
I will have to double check if that is expected behavior or if it is a bug. For now I am labeling as both a bug and a proposal

LuxoftAKutsan, 2017-12-01:
Priority is low
SDL send not critical additional notification

jacobkeeler, 2018-03-08:
Root cause: https://github.com/smartdevicelink/sdl_core/blob/master/src/components/application_manager/src/commands/mobile/unregister_app_interface_request.cc#L51

joeljfischer, 2019-04-08:
This has become a bigger issue with the iOS library recently due to Core sending the `-1` bad data. Ideally Core should never be sending data that doesn't match the type that's expected.

HSavynetska, 2019-11-19:
The defect is reproduced in the current develop branch

Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/2109
SDL build: https://github.com/smartdevicelink/sdl_core/tree/develop ( 891497b )
ATF: develop branch

E-SAITO-TMC, 2020-05-13:
@theresalech -san, We were able to reproduce this issue with Core 6.0.1, This is a medium priority issue for Toyota. Could you please include this in the Core 6.2 release planning?

iCollin, 2020-10-01:
Closed via https://github.com/smartdevicelink/sdl_core/pull/3389

PR #3389 (2020-10-01): Fix problem with sending onAppInterfaceUnregistered notification during execution of UnregisterAppInterface request
Fixes #1178 

This PR is **ready** for review.

### Risk
This PR makes **no** API changes.

### Testing Plan
Corresponding unit test is updated

### Summary
This pull request is intended to fix issue with sending onAppInterfaceUnregistered notification during execution of UnregisterAppInterface request. Root cause of this issue is a sending OnAppInterfaceUnregistered notification with invalid unregistration reason in method UnregisterAppInterfaceRequest::Run().
That's why this no [... cut]
closed 2020-10-01 by iCollin