smartdevicelink/sdl_core #1206: HTTP: SDL doesn't send SDL.OnStatusUpdate(UPDATE_NEEDED) in case of failed retry strategy during previous IGN_ON

KhrystynaDubovyk, 2017-01-24:
# Description
## Steps to reproduce
*Required steps for testing*
 Run script [004_ATF_P_TC_REQUEST_PTU_Trigger_PTU_failed_previous_IGN_ON_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/004_ATF_P_TC_REQUEST_PTU_Trigger_PTU_failed_previous_IGN_ON_HTTP.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-05-09:
PR: #1487

LuxoftAKutsan, 2017-12-01:
Priority is high
Significant loss of functionality on SDL side with the standard use-case.

LuxoftAKutsan, 2017-12-06:
@jacobkeeler I can see PR related to this PR. Shouldn't this issue be closed?

LuxoftAKutsan, 2017-12-06:
Not reproduced. Checked with script https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/4_5/1206_REQUEST_PTU_Trigger_PTU_failed_previous_IGN_ON.lua 
Script merged with PR : https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1752/files

dboltovskyi, 2017-12-11:
Please note issue was checked on current develop (https://github.com/smartdevicelink/sdl_core/commit/9e95f67178c575ca5a537f9ef3a69653219b073d)

dboltovskyi, 2017-12-11:
For the information: issue was reproducible on https://github.com/smartdevicelink/sdl_core/commit/8dc5c15 commit

LuxoftAKutsan, 2017-12-14:
wrong switching between update states. Fix available and already merged to develop with https://github.com/smartdevicelink/sdl_core/pull/1487/

mrapitis, 2017-12-21:
Reviewed ATF script and ran successfully. Successfully reproduced issue with provided commit.