smartdevicelink/sdl_core #1207: DISALLOWED instead of USER_DISALLOWED in case getting negative user consent from HMI

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [App_Permissions/174_ATF_HP_User_Consent_NO.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/App_Permissions/174_ATF_HP_User_Consent_NO.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-05-08:
Issue is not reproducible or it is fixed in develop.

JackLivio, 2017-05-08:
@KhrystynaDubovyk the title of the issue and the actual/expected outcomes seem to be conflicting. What is the correct behavior supposed to be in this case?

JackLivio, 2017-05-08:
When testing Develop manually, mobile is receiving "DISALLOWED"

dboltovskyi, 2017-05-10:
@JackLivio, Thank you for such finding, - actual and expected results were confused.
Description is updated and now it is correct.

LuxoftAKutsan, 2017-12-01:
Priority is Medium
Wrong result code received from SDL

GetmanetsIrina, 2018-07-03:
Defect is not reproduced.
SDL: develop branch(ba1da7bc2ff3fd37638620f6bea526384646ab0e)
ATF: develop branch
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/App_Permissions/004_ATF_HP_User_Consent_NO.lua

closed 2018-07-03 by GetmanetsIrina