smartdevicelink/sdl_core #1208: SDL sends BC.ActivateApp twice in specific case

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [202_ATF_OnAllowSDLFunctionality_allowed_true_without_device.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Related_HMI_API/202_ATF_OnAllowSDLFunctionality_allowed_true_without_device.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

LuxoftAKutsan, 2017-12-01:
Priority is low
Redundant not critical SDL activity

APipko, 2018-08-15:
sdl_core commit 406f491b7db0123351e3851d130bb334cf275023 : not reproduced

IGapchuk, 2018-10-01:
SDL (develop): commit - 28235f88a6733106a7e8a0d490dfb16a5f374a33.
ATF: (feature/script_runner): commit - 6d7161a86825dd75ae5ae9ac6bae9d9ee824880e.
Status: not reproduce.

iCollin, 2021-03-31:
Closed as not reproducible.
Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin