smartdevicelink/sdl_core #1209: Second device is not connected

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [ATF_For_PTU_use_consented_device_only.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/ATF_For_PTU_use_consented_device_only.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

yangbiao01, 2017-02-22:
The mobile appname and appid don't be same

jacobkeeler, 2017-02-22:
@DrachenkoAnastasiia I assume that step 5
> 5. Deactivate application #2 

was a typo?  I don't see anything in the script which deactivates the second application prior to performing the PTU, so I gather that this was just supposed to be the activation step rather than the deactivation step.

Also, you should probably include information stating that the first device is allowed SDL functionality and the second isn't (as part of steps or preconditions), as this is an important part of the scenario.

LuxoftAKutsan, 2017-12-01:
Priority is medium
Loss of functionality is specific use case

robinmk, 2018-05-16:
@LuxoftAKutsan , Please provide clarity on the use case. 
The description of the issue is not clear. Please also respond to Jacob's comment.

aderiabin, 2018-05-17:
Current defect was created with supposition of existence of SDL floating defect, which was reproduced  time to time on CI. 
Deep investigation of the defect led to next results: defect is not reproduced during continuous runs of the test script in local test environment nor in current CI server environment (which was significant improved comparing with CI environment when defect was created ), SDL works as designed.
SDL commit: 65c1e7e (current develop)

I propose to close the defect as not actual.

GetmanetsIrina, 2018-06-14:
Defect is not reproduced on [develop branch c7f493a5](https://github.com/smartdevicelink/sdl_atf_test_scripts/commit/c7f493a50661f9a20ea242b1338f0f19dd21e52c).

closed 2018-06-14 by GetmanetsIrina