smartdevicelink/sdl_core #1210: PROPRIETARY: BC.PolicyUpdate is sent during retry sequence

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [ATF_Start_PTU_retry_sequence.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/ATF_Start_PTU_retry_sequence.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-03-14:
It appears that this OnStatusUpdate message is sent as part of the `OnAppRegisteredOnMobile` function (eventually leading to the `OnUpdateStatusChanged`, where this message is sent from). 

This function call was moved in this commit (although it was part of the `RegisterAppInterface` response originally): https://github.com/smartdevicelink/sdl_core/commit/088bc6dbd4a59b4bcea423accef081cb7beae889

Itileda, 2017-03-15:
Actually SDL sends OnStatusUpdate(UPDATE_NEEDED) to HMI as expected, but issue here is that there is a redundant PolicyUpdate request to HMI.
Initially snapshot was created during first BasicCommunication.PolicyUpdate.
Accordinf to retry sequence diagram, in case mobile App doesn't reply within timeout, SDL should inform HMI that status was changed from UPDATING to UPDATE_NEEDED, and send again to the App OnSystemRequest. Next behavior is the same. Right after sending OnSystemRequest state is changed to UPDATING.

jacobkeeler, 2017-03-15:
@Itileda I don't know how this comment ended up here, it was originally posted on #1215, and for some reason a duplicate was posted here as well. Sorry about that.

jacobkeeler, 2017-03-20:
@Itileda Could you please link the retry sequence diagram? I'm not finding it on the Dev Portal. 

Based off of the "Proprietary flow" diagram for [PolicyUpdate](https://smartdevicelink.com/en/docs/hmi/master/basiccommunication/policyupdate/), this message is supposed to start off the PTU, is this not supposed to initiate the sequence on retry?

LuxoftAKutsan, 2017-12-01:
Priority is low.
Redundant SDL activity that may trigger significant effect on HMI

jacobkeeler, 2020-04-20:
Closing with the merge of #3329

closed 2020-04-20 by jacobkeeler