smartdevicelink/sdl_core #1211: SDL shut down due to SIGSEGV

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [Check_count_of_removals_for_bad_behavior_REQUEST_WHILE_IN_NONE_HMI_LEVEL.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Validation_of_PolicyTables/Check_count_of_removals_for_bad_behavior_REQUEST_WHILE_IN_NONE_HMI_LEVEL.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-05-10:
PR: #1519

LuxoftAKutsan, 2017-12-01:
@jacobkeeler @JackLivio I can see merged PR related to this issue. Should it be closed?

LuxoftAKutsan, 2017-12-01:
Priority is high
Segmentation fault in a trivial use case

LitvinenkoIra, 2017-12-06:
Works correctly on current develop (9e95f67178c575ca5a537f9ef3a69653219b073d)
Script for verifying and testing report [1211.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1535131/1211.tar.gz)

dboltovskyi, 2017-12-11:
For the information: issue was reproducible on https://github.com/smartdevicelink/sdl_core/commit/22ed9b1 commit

LuxoftAKutsan, 2017-12-14:
Issue was in invalidation iterator of the application during checking allowed parameters. Fix available in  https://github.com/smartdevicelink/sdl_core/pull/1519

mrapitis, 2017-12-21:
Reviewed ATF script and ran successfully.  Successfully reproduced issue with provided commit.

jacobkeeler, 2018-01-08:
Fixed with the merge of #1519

closed 2018-01-08 by jacobkeeler