smartdevicelink/sdl_core #1212: Application is unregistered due to TOO_MANY_REQUESTS

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [292_ATF_P_Memory_Kb_Constraints_Ignoring.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Validation_of_PolicyTables/292_ATF_P_Memory_Kb_Constraints_Ignoring.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

JackLivio, 2017-05-08:
I was not able to reproduce this issue

Is this how I was supposed to define pre_DataConsent in the preloaded pt?
```
            "pre_DataConsent": {
                "keep_context": false,
                "steal_focus": false,
                "priority": "NONE",
                "default_hmi": "NONE",
                "groups": ["BaseBeforeDataConsent"],
                "memory_kb": 200
            }

```

DrachenkoAnastasiia, 2017-05-10:
@Itileda, please, take care about this question.

jacobkeeler, 2017-05-10:
PR: #1513

LuxoftAKutsan, 2017-12-01:
@jacobkeeler @JackLivio I can see merged PR related to this issue. Should it be closed?

LuxoftAKutsan, 2017-12-01:
Priority is hight
Loss of functionality. Wrong result code returned from SDL

JackLivio, 2018-03-01:
Fixed by #1513 and #1486

closed 2018-03-01 by JackLivio