smartdevicelink/sdl_core #1213: SDL does not increment "count_of_sync_reboots" section value of Local Policy Table on getting SDL.OnSystemError("SYNC_REBOOTED")

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [074_ATF_usage_and_error_counts_update_count_of_sync_reboots.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Validation_of_PolicyTables/074_ATF_usage_and_error_counts_update_count_of_sync_reboots.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

LuxoftAKutsan, 2017-12-01:
Priority is low
Sometimes SDL does not increment an internal counter.

iCollin, 2021-03-31:
Closed as not reproducible against the first 7.1 release candidate. The mentioned ATF test has been renamed (228) but it is passing.

Please let me know if you believe this issue is still actual.

closed 2021-03-31 by iCollin