smartdevicelink/sdl_core #1214: UPDATE_NEEDED is sent if user requested PTU from HMI

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [007_ATF_P_TC_OnStatusUpdate_UPDATE_NEEDED_new_PTU_request.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/007_ATF_P_TC_OnStatusUpdate_UPDATE_NEEDED_new_PTU_request.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-05-09:
PR: #1491

LuxoftAKutsan, 2017-12-01:
Priority is low.
Redundant SDL activity. But this activity may trigger significant effects on HMI.
@jacobkeeler @JackLivio I can see merged PR related to this issue. Should it be closed?

JackLivio, 2018-03-01:
Yes issue is fixed by pr #1491. Closing issue

closed 2018-03-01 by JackLivio