smartdevicelink/sdl_core #1215: UPDATE_NEEDED is sent in case of unconsented device

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [018_ATF_P_TC_Sending_PTS_to_mobile_application.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/Policy_Table_Update/018_ATF_P_TC_Sending_PTS_to_mobile_application.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-03-14:
It appears that this `OnStatusUpdate` message is sent as part of the `OnAppRegisteredOnMobile` function (eventually leading to the `OnUpdateStatusChanged` function, where this message is sent from). 

This function call was moved in this commit (although it was part of the `RegisterAppInterface` response originally, not the `ActivateApp` request): https://github.com/smartdevicelink/sdl_core/commit/088bc6dbd4a59b4bcea423accef081cb7beae889

Itileda, 2017-03-15:
There is a **requirement:**
"PoliciesManager must initiate the PT Update in case the app that has never received the updated policies registers from non-consented device and then the User consents this device.

**Note:**
1. This requirement applicable ONLY for EXTENDED_PROPRIETARY flow"

**Verification criteria:**
Precondition:
a. SDL and HMI are running
b. Local PT does not have user_consent_records for <Device> (device is not yet consented)
c. <Device> is connected, <appID> is running on that device
Action:
app->SDL: RegisterAppInterface
Expected:
SDL->HMI: OnAppRegistered (appID)
PoliciesManager: device needs consent; app is not present in Local PT.
PoliciesManager: "<appID>": "pre_DataConsent" //that is, adds appID to the Local PT and assigns default policies to it
HMI->SDL: SDL.ActivateApp (appID)
SDL->HMI: SDL.ActivateApp (isSDLAllowed:false, device)
HMI->SDL: SDL.GetUserFriendlyMessages ("DataConsent")
SDL->HMI: SDL.GetUserFriendlyMessages_response
HMI displays the device consent pormpt. User consents the device
HMI->SDL: OnAllowSDLFunctionality(isSDLAllowed:true)
SDL->HMI: OnStatusUpdate (UPDATE_NEEDED)//PTU sequence started
PTS is created by SDL:
SDL-> HMI: BC.PolicyUpdate()

LuxoftAKutsan, 2017-12-01:
Priority is medium.
Redundant sdl activity. But this activity may trigger significant effects on HMI.

LuxoftAKutsan, 2017-12-01:
@jacobkeeler @JackLivio I can see merged PR related to this issue. Should it be closed?

JackLivio, 2018-03-01:
Fixed by #1474

closed 2018-03-01 by JackLivio