smartdevicelink/sdl_core #1216: HTTP: OnSystemRequest is sent twice

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [025_ATF_P_TC_PTU_Local_PT_Start_Retry_Sequence_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/025_ATF_P_TC_PTU_Local_PT_Start_Retry_Sequence_HTTP.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

LitvinenkoIra, 2017-12-01:
Priority : Medium
Redundant request to mobile

JackLivio, 2018-03-01:
Issue no longer reproducible. Closing

closed 2018-03-01 by JackLivio