smartdevicelink/sdl_core #1217: OnHMIStatus is not sent in case app gets 'null' while PTU

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [162_ATF_P_TC_HMI_Status_Appid_Gets_Null_In_Case_Of_PTU.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/appID_Management/162_ATF_P_TC_HMI_Status_Appid_Gets_Null_In_Case_Of_PTU.lua)
Run script [163_ATF_P_TC_HMI_Status_Value_Of_AppId_In_PT_Is_Null.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/appID_Management/163_ATF_P_TC_HMI_Status_Value_Of_AppId_In_PT_Is_Null.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-05-10:
PR: #1513

LitvinenkoIra, 2017-12-01:
@jacobkeeler is this issue shouldn't be closed? I see PR related to this Issue.

LitvinenkoIra, 2017-12-01:
Priority : minor
Not critical redundant notification to mobile

JackLivio, 2018-03-01:
Fixed by #1513

closed 2018-03-01 by JackLivio