smartdevicelink/sdl_core #1218: Mandatory parameter 'spaceAvailable' is not provided in response for PutFile request

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [158_ATF_P_TC_Register_App_Interface_With_Data_Consent_Assign_Default_Policies.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/appID_Management/158_ATF_P_TC_Register_App_Interface_With_Data_Consent_Assign_Default_Policies.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-02-13:
While it should likely be included in this case, it seems to me that the spaceAvailable parameter shouldn't be mandatory in the first place. 

Leaving this as mandatory requires it to be included even when sending an error response, which theoretically shouldn't include anything other than the `success`, `resultCode`, and possibly `info` parameters.

jacobkeeler, 2017-03-16:
@DrachenkoAnastasiia Also, when I run the test myself, spaceAvailable is not missing in the SUCCESS case, just the DISALLOWED case. As I said before, this parameter should not be mandatory in error cases, but the API treats it as such.

AStasiuk, 2017-03-17:
@jacobkeeler Earlier all changes in MOBILE_API.xml were threated as change requests. 
Need to clarify how such cases should be handled?For example to create sdl_evolution proposal for future discussion, or some other approach.
Looking forward you suggestion.

LitvinenkoIra, 2017-12-01:
@jacobkeeler should we create proposal related to it?

LitvinenkoIra, 2017-12-01:
Priority is High
Missed mandatory parameters

LitvinenkoIra, 2017-12-05:
Works correctly on current develop (8c94933)
Script for verifying and test reports in attachments.
[1218.tar.gz](https://github.com/smartdevicelink/sdl_core/files/1530894/1218.tar.gz)

LitvinenkoIra, 2017-12-05:
@jacobkeeler @JackLivio should we create proposal related to it? or just close this issue?

JackLivio, 2018-03-01:
Leaving issue open until proposal is made for making parameter non mandatory in failed response

HSavynetska, 2018-06-14:
Defect is not reproduce on current develop(91123c3cae83ad41e1f5f37d37e46e9f07264d1b)

Verified with script(Script under review ): https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1923

AStasiuk, 2018-06-15:
Closed as not reproducible.

closed 2018-06-15 by AStasiuk

GetmanetsIrina, 2018-07-30:
Script for reproducing https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Defects/5_0/1218_Mandatory_parameter_spaceAvailable_is_not_provided_in_response_for_PutFile_request.lua