smartdevicelink/sdl_core #1219: HTTP: SDL use incorrect URL while sending OnSystemRequest to application

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [011_ATF_P_TC_PTS_Define_URL_to_send_PTS_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/011_ATF_P_TC_PTS_Define_URL_to_send_PTS_HTTP.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

LitvinenkoIra, 2017-12-01:
Priority is Medium 
Functionality was not broken, but wrong application chosen.

yang1070, 2018-09-20:
@DrachenkoAnastasiia @ZhdanovP
I don't understand step 4 in the precondition.
```
Make sure LPT has different urls for app_1 and app_2 in endpoints section
Ex.:
app_1: http://policies.domain1.com/api/policies
app_2: http://policies.domain2.com/api/policies
```
I know in policy, in `module_config`, there is a `endpoints`, for example
```
            "endpoints": {
                "0x07": {
                    "default": ["http://x.x.x.x:3000/api/1/policies/proprietary"]
                },
                "0x04": {
                    "default": ["http://x.x.x.x:3000/api/1/softwareUpdate"]
                },
                "queryAppsUrl": {
                    "default": ["http://sdl.shaid.server"]
                },
                "lock_screen_icon_url": {
                    "default": ["http://i.imgur.com/TgkvOIZ.png"]
                }
            },
``` 
But that applies to all the apps. How come each app has its own `endpoints` defined?

dboltovskyi, 2018-09-26:
@yang1070 
It's possible to define URL for each app in a following way:
```
"endpoints": {
                "0x07": {
                    "0000001": [
                        "http://policies.domain1.com/api/policies"
                    ],
                    "0000002": [
                        "http://policies.domain2.com/api/policies"
                    ],
                    "default": [
                        "http://x.x.x.x:3000/api/1/policies/proprietary"
                    ]
                }
            }
```

jacobkeeler, 2020-05-07:
Fixed via #3329

closed 2020-05-07 by jacobkeeler