smartdevicelink/sdl_core #1220: HTTP: SDL omits one of the URLs while cyclically changing during retry sequence

DrachenkoAnastasiia, 2017-01-24:
## Steps to reproduce:
Run script [ATF_P_TC_PTU_Cycleing_Through_The_URLs_During_Retry_Sequence.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/ATF_P_TC_PTU_Cycleing_Through_The_URLs_During_Retry_Sequence.lua)

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery
- [ ] Source code updates
- [ ] Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-02-22:
@DrachenkoAnastasiia Which URLs are omitted in this example case?

LuxoftAKutsan, 2017-02-23:
@jacobkeeler  As I can see from logs ( attached) : 
<ErrorMessage Status="FAILD">{ ['AutoFail: 1'] = '
    Expected url:
        http://policies.domain1.ford.com/api/policies
    Actual:
        http://policies.domain2.ford.com/api/policies
',}</ErrorMessage>

And script contents :

local r_expected = {
  "http://policies.telematics.ford.com/api/policies",
  "http://policies.domain1.ford.com/api/policies",
  "http://policies.domain2.ford.com/api/policies",
  "http://policies.domain3.ford.com/api/policies",
"http://policies.domain4.ford.com/api/policies"}

url : http://policies.domain1.ford.com/api/policies missed 
[logs2.zip](https://github.com/smartdevicelink/sdl_core/files/796139/logs2.zip)

jacobkeeler, 2017-05-09:
PR: #1478

LitvinenkoIra, 2017-12-01:
@jacobkeeler  shouldn't this issue be closed? I can see PR this fixes it. 
Priority is Low : loss of insignificant functionality
SDL skip one of the url, but functionality works.

GetmanetsIrina, 2018-06-19:
Environment:
Script: https://github.com/smartdevicelink/sdl_atf_test_scripts/pull/1932
SDL build: develop branch (90ee6ad4af35d9fdbc00df6c72a8fb8805f3c5da)
ATF: develop branch

Status: defect is not reproduced.

closed 2018-06-19 by GetmanetsIrina