smartdevicelink/sdl_core #1221: HTTP SDL doesn't respect retry timeout sequence

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [032_ATF_P_TC_PTU_Retry_Sequence_Retry_Timeout_Expiration_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/032_ATF_P_TC_PTU_Retry_Sequence_Retry_Timeout_Expiration_HTTP.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-05-08:
PR: #1483

LitvinenkoIra, 2017-12-01:
@jacobkeeler shouldn't this issue be closed? I can see merged PR that is related to it.

LitvinenkoIra, 2017-12-01:
Priority Medium
Loss of  functionality in standard use case. 
Retry timeouts should be increased between retries

JackLivio, 2018-03-01:
fixed by #1483

closed 2018-03-01 by JackLivio