smartdevicelink/sdl_core #1222: HTTP: Section consumer_friendly_messages.messages updated partially

KhrystynaDubovyk, 2017-01-24:
## Steps to reproduce
*Required steps for testing*
 Run script [046_ATF_P_TC_PTU_UTF8_Encoding_Check_HTTP.lua](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/feature/external_proprietary_policy/test_scripts/Polices/build_options/046_ATF_P_TC_PTU_UTF8_Encoding_Check_HTTP.lua) 

## Actual result
Scenario is failed

## Expected result
Scenario is passed

## Environment
repo: https://github.com/smartdevicelink/sdl_core/
branch: feature/external_proprietary_policy (head of branch)
ATF version: 2.2, commit 752d9aec810ccb2a30ab6b53c587f8bb8ef1c545

## Expected delivery

- [ ]  Source code updates
- [ ]  Code comments
- [ ]  UTs add/update
- [ ]  ATF tests add/update
- [ ]  Add/update CI plans/jobs
- [ ]  SDD updates
- [ ]  Guidelines update (sdl_core_guides)
- [ ]  Guidelines update (sdl_hmi_integration_guidelines)

jacobkeeler, 2017-03-20:
@KhrystynaDubovyk Does this only occur with the "HTTP" mode?

LitvinenkoIra, 2017-12-01:
Priority Medium.
Loss of functionality in non-standard use case.

LitvinenkoIra, 2019-12-11:
According to the description in https://github.com/smartdevicelink/sdl_core/pull/2295 - to reproduce this issue use the following ATF test script - 
ATF script: [083_ATF_PTU_UTF8_Encoding_Check_HTTP](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/083_ATF_PTU_UTF8_Encoding_Check_HTTP.lua)

LitvinenkoIra, 2019-12-11:
[083_ATF_PTU_UTF8_Encoding_Check_HTTP](https://github.com/smartdevicelink/sdl_atf_test_scripts/blob/develop/test_scripts/Policies/build_options/083_ATF_PTU_UTF8_Encoding_Check_HTTP.lua) - successfully passed on the latest develop (e1375565)
**Defect is not reproducible** because this issue was fixed in https://github.com/smartdevicelink/sdl_core/pull/2783/commits/2c83332223342252286f7d33520b83070259ef20 and was merged into develop as a part of [Feature/SDL passenger mode](https://github.com/smartdevicelink/sdl_core/pull/2783)

closed 2019-12-11 by LitvinenkoIra